ev3dev/ev3dev #300: Best way to support multiple Ev3's

sharived, 2015-03-26:
I now have 3 Ev3's running Ev3Dev and I'm trying to figure out the best way to work with all of them simultaneously.  I assume I need to have different IP addresses setup and was wondering what the best way to do this.  I've figured out how to change the hostname, but I've never been able to get the hostname recognized by my PC when I try and ssh into it.  So if you know how to be able to ssh using the hostname instead of the ip address that would be great as well .  BTW, my host is typically a Mac and I use bluetooth (untethered).

moriarty, 2015-03-27:
To ssh into a machine by hostname requires dns. Some routers provide this as machines connect to the network, but not all do. But with Bluetooth, the router and dns server isn't involved. 

The quickest solution if you need to access your EV3s by hostname, would be to hard-code them directly into the /etc/hosts file on your mac. This is a system wide change on your mac, and requires sudo. But it is useful if you're configuring programs which access the EV3, and you want to write config files with hostnames not ip addresses.

But, if you only want the convience of typing the ssh command with hostname, then the better option is to set up your ssh config file. It's "~/.ssh/config" 

I'm in the office and not on my mac, nor is this computer configured with my EV3, but I'll provide the two files as examples.

/etc/hosts

    #
    # /etc/hosts: static lookup table for host names
    #

    #<ip-address>	<hostname.domain.org>	<hostname>
    127.0.0.1	localhost.localdomain	localhost moriarty-xmg
    ::1		localhost.localdomain	localhost moriarty-xmg

    192.168.1.100	refbox
    192.168.1.223	workstation
    192.168.1.101	conveyor
    192.168.1.102	camera
    192.168.1.103	drill
    192.168.1.210	youbot

    # End of file

~/.ssh/config

    Host rockin
        hostname    refbox
        user        rockin
        ForwardX11  yes

    Host conveyor
        hostname    conveyor
        user        pi

    Host workstation
        hostname    workstation
        user        alex
        ForwardX11  yes

     Host yb
         hostname youbot-brsu-2-pc-1
         user    alex

With the above example, if I run the command 
    $ ssh rockin
it's the same as if I had typed
    $ ssh -X rockin@192.168.1.100

I don't know if it's useful for you, but if you're switching between networks often, bluetooth, wireless, wired, I find it helpful to distinguish the different hostnames. That way you know which interface you're using. The robot I usually use has 3 network interfaces, one wireless using dhcp, one ethernet using dhcp, and one ethernet using static ip.

sharived, 2015-03-27:
Thanks for the detailed response!  I think this will work for my use, however I still need to come up with a solution for my customers.  I want to be able to support a classroom full of kids running Ev3Dev and I want it super simple to use.  I'll be supplying SD cards with my product (camera) with everything setup.  If someone orders more than 1 kit then I want to have them use different IP addresses and hostnames.  I can't make changes to their PC's.  So maybe I should just have different IP addresses set up and avoid ssh'ing altogether.  I'll give that a try.  So I guess I just need to figure out how to set up different IP addresses for each Ev3.  Any suggestions?

G33kDude, 2015-03-27:
I've found installing the samba host name integration packages to work pretty well for connecting via host name from a windows computer. I'm not sure if it'd work for connecting from a mac though. I'm a bit rough on the details, but if I remember correctly, installing "libnss-winbind" did the trick. This won't allow for pinging out from the EV3 by hostname, that requires a bit of fiddling in `/etc/nsswitch.conf` last I checked. I'll be reformatting my EV3 today, so I can tinker around with it a bit and get back to you.

moriarty, 2015-03-27:
Unfortunately there is no one solution that can be gaurenteed to work for all cases without setting up the IP addresses manually and configuring the users PCs. 

We have the same situation with in a university lab setting, with a bunch of Kuka YouBots, other robots, networked devices (raspberry pis from my config files above). We're always looking for an easier way to set everything up, less hardcoded and more automagically. 
We're not using bluetooth or tethering but I'll give my 2cents here. 

The easiest solution, which I use with my two EV3s at home and the YouBots at work, is to connect them via wifi and use the DHCP from the router to provide different addresses. Then to display the address on the device screen. 
-- But this doesn't guarentee you'll get the same IP address every time you connect, to do that you need access to the router menu and they usually have some way of clicking on the currently connected devices and setting the IP to always be the same. 

This isn't always straight forward in educational institutes were there is some network administrator controlling access to the routers. And it also dependent on the router. And if for some reason a classroom is being used for exams, and you need to move the robots to another classroom, the other classroom will not have the router configured properly. 
-- If this is a problem, the easiest solution we've found is buy routers. Which does sound like a terrible solution, but it is what's being done. 

I've participated in the past 3 RoboCup World Championships, networking is always an issue. Even with Cisco as a event sponsor, the problem is all of the teams are dependent on their network configuration. Imagine, 400 wifi networks with over 3000 devices in one or two buildings. In the RoCKIn EU competition, we assign static IP addresses for each robot, and one for the team to connect to their robot, and then we let them bring their own routers  but we force them to all use only one wireless channel, leaving the other non-overlapping channels available for the official networks. 

In our univesity, we're using high end commercial and non-commercial routers, with additional antennas, plugged in to the existing university infrustructure so that we have control over the networks which the robots connect to. Overkill for most situations. But we need to compete with the wireless communcations guys who share our lab space, and we want to be sure that we're able to connect at events like RoboCup.

Long story short: 
IMHO, a router + USB adapters would be the simplest solution. 
But unfortunately it would add a cost of approximately 20+10*(number of ev3s) for a classroom set. We keep some USB powered TP Link travel routers on hand, they're small and quick and easy to set up for demos or hacking in hotels/airports, and currently on sale for <20 on amazon.com

G33kDude, 2015-03-27:
It may be worth noting that it's possible to set up many normal wifi devices as access points. I've done this with my laptop before when using the EV3 in a public setting where I didn't have access to a network. I couldn't get it to work using ad-hoc networking, but I was able to set up a secondary virtual wifi network using the new "Microsoft Virtual WiFi Miniport Adapter" feature of windows 7. Most smartphones are able to do something similar last I checked.

dlech, 2015-03-27:
It is possible to do DHCP over Bluetooth (or USB). It is just temperamental, which is why we recommend using a static IP. It works if you have Internet connection sharing enabled on your host computer (OS X or Windows). On the EV3, on the IPv4 tab, select "Change..." and then "Use DHCP" before connecting.

If it doesn't connect, disabling and enabling Internet connection sharing on the computer and repairing Bluetooth. I have to do this quite a bit, which is why I said it was temperamental.

On my Mac, I can see the EV3 host names by running `cat /private/var/db/dhcpd_leases`, but I don't think that the dns resolver takes this into consideration. I haven't switched back to windows to see if it works.

The ideal situation would be if we could enable bonjour/zerconf on the EV3 for connecting to OS X (this would allow the host computer to find the ev3 by hostname) and enable `libnss-winbind` for connecting to Windows. But, this will take some doing to figure out.

At the same time, if you only have one EV3 per host computer and you are connecting via Bluetooth or USB (assuming that this will be the case in the classroom), then all of the EV3s can have the same hostname and IP address with no problem. The Bluetooth connections are all isolated, so they won't be aware that there is another device with the same address. Although, now that I am thinking about it, Bluetooth uses the hostname for the Bluetooth name as well, which could be a problem trying to pair the correct EV3.

And finally, as to an easy way to change the hostname on the EV3... This should be fairly easy to add to brickman using systemd. From the command line `hostnamectl set-hostname NAME`. In brickman though, we should use D-Bus (http://www.freedesktop.org/wiki/Software/systemd/hostnamed/). Please open a new issue for this item if you think it is useful.

sharived, 2015-03-27:
I have been using DHCP w/ Bluetooth for a while without any issues - didn't realize it was experimental.  However I think I'm better off with a static IP, otherwise I'm getting different IP address on different host computers.  Is there a way for me to store the address like you do for the Mac/Windows presets instead of entering the address manually?  Would be best if I could just program this on bootup.

dlech, 2015-03-27:
Its not "experimental", just "temperamental". And really it is just Internet Connection Sharing on Windows. If you are using Mac, then you probably won't have as many issues.

Not sure what you mean by "the Mac/Windows presets", but once you specify a manual address on the EV3, it will remember it across reboots.

sharived, 2015-03-27:
By presets I meant the menu that has "Load Mac Defaults" for the IPv4 change menu.  I was hoping to store the IP address somewhere instead of having to enter through the interface.  For example if I have to do this 200 times (wishful thinking) looking for a quick way to do this.

dlech, 2015-03-27:
Why not just use one of the existing defaults? Or, you could write a program to do this, for example using connmanctl.

sharived, 2015-03-31:
someone recommended I try avahi - do you think this would work on Ev3?

dlech, 2015-03-31:
Avahi is the "bonjour/zerconf" that I mentioned earlier. I don't see any reason why it should not work. I am pretty sure that you have to install extra software on Windows to make it work with Windows though.

dlech, 2015-03-31:
See https://github.com/ev3dev/ev3dev/issues/103#issuecomment-49368237

sharived, 2015-04-14:
I've been using avahi and it works great.  Haven't attempted to try on windows yet, but great on Mac.  Closing issue.

closed 2015-04-14 by sharived

dlech, 2015-04-14:
Great! Do you mind sharing how you set it up?

sharived, 2015-04-15:
Sure.
I did the installation
 apt-get install avahi-daemon libnss-mdns

Then I changed the hostname for each device
hostnamectl set-hostname ev3dev0
hostnamectl set-hostname ev3dev1

I'm using Snap! so I can connect
to to ev3dev0.local and ev3dev1.local

I connect via bluetooth.