ev3dev/ev3dev #301: Why does the Ev3 sometimes come up without brickman display?

sharived, 2015-03-26:
I had cloned my SD card and the first time I tried it, I got the login screen instead of the Brickman display.  I re-cloned to another SD card and it worked!  Now with the same SD card I'm back to the login screen.  I was wondering what causes this - I had never encountered it before - and how to keep this from happening.

Thanks,
Shari

sharived, 2015-03-26:
Now that same SD card won't get past the Ev3Dev booting screen...

sharived, 2015-03-27:
Any suggestion for this,  should I try another SD card or do you think I have a bad image?

dlech, 2015-03-27:
See #267. I think that part of the old image is getting left around and since the new image doesn't write over the entire SD card, it can cause problems.

I think the easiest thing to do right now is use `dd` to write all 0s to your SD card to wipe it clean and then flash the image again.

sharived, 2015-03-27:
That did the trick!  Thanks!

dlech, 2015-03-27:
Great. We should have this sorted out in the next image release.

sharived, 2015-04-14:
I think something strange is still going on here.  I have the issue where the SD card is working fine - brickman comes up fine and then I bring it up again, and I'm getting the login prompt instead of brickman.  Do you still think this is the same issue or is something else going on?  I had 0'd out the SD card before writing it.  I only have 1 brand of SD card at the moment, so that could still be the problem - It's a Kingston 4GB card.  I will try buying something else and see if the problem goes away.

dlech, 2015-04-14:
> I have the issue where the SD card is working fine - brickman comes up fine and then I bring it up again,

By "bring it up again", do you mean reboot?

> Do you still think this is the same issue or is something else going on?

I think this is a different/new issue. Do you have additional services enabled (such as avahi that you mentioned in another issue) that could make the boot time longer? Also, does brickman stop or crash or is it just running on a different virtual console? Which virtual console is the unwanted login screen running on?

sharived, 2015-04-15:
In this case, I hadn't booted the device in a while (it had been working fine) and when I turned it on it was on the login screen.

I hadn't installed Avahi yet.  I am starting my own service that runs my python application on startup.  

It says tty1, is that what you mean?  I checked and didn't see any instances of brickman running.
I tried starting it with /usr/sbin/brickman and got the error - "error while loading shared libraries: /usr/lib/arm[linux-gnueabi/libgio-2.0.s0.0: Unable to run arch specific checks"

I'm doing another update and will try again.  Not sure I have all the latest code - last updated a couple of weeks ago.

dlech, 2015-04-15:
> It says tty1, is that what you mean?

Yes.

systemd will automatically attach a login screen to any unused virtual console. It sounds like brickman is failing to start because of this error, so instead of brickman getting tty1, systemd sees that nothing is using it and you get the login screen.

I vaguely remember seeing the error before, but I am not recalling what I did about it. Hopefully upgrading everything will take care of it.

sharived, 2015-04-15:
I think it is my SD cards.  I have 5 of the same brand - these were cards I had sitting around.  So far 2 are working and 2 I have thrown away - they worked, stopped working and then when I try and zero them out again I get an error.  I'm trying to find some new ones.  I know that you guys recommend class 10 - so far having trouble finding smaller (ie cheaper) class 10 cards.  If there is a particular brand you'd recommend let me know.

dlech, 2015-04-15:
> when I try and zero them out again I get an error.

What is the error message?

> I know that you guys recommend class 10

Not exactly. I've said it before: The class only tells you the minimum *sequential* read/write speed. This is good for cameras that only do sequential read/writes. But what we are doing mostly is random access. You have to dig to find those numbers. In fact, from what I have read, some class 10 cards have way worse random read/write speeds than class 4 or 6 cards.

This is an interesting read: <http://forums.whirlpool.net.au/forum-replies.cfm?t=1582172>. Just read the first post and then skip to the last page.

I have a SanDisk Extreme Pro 8GB that is noticeably faster than my old 2GB generic cards. Still, the old cards are not unbearably slow.

It sounds like the most import thing is to buy from a reputable distributor so you don't get a knock-off.

[newegg](http://www.newegg.com/Product/ProductList.aspx?Submit=ENE&N=100007962+600006221+600006225+600006209+600006226+600006195+4024&IsNodeId=1&Description=microsd&bop=And&Order=PRICE&PageSize=30&Page=1) has gobs of cards for under $10. Unfortunately, none of them list random access read/write speeds.

dlech, 2015-05-21:
In the latest image (ev3-ev3dev-2015-05-20), I have changed some boot order dependencies which will hopefully prevent this from happening. Also, now brickman (v0.6.1) will always run on tty1.

Please re-open the issue if you still have this problem.

closed 2015-05-21 by dlech