ev3dev/ev3dev #302: Mmap for interacting with the hardware doesn't work?

theZiz, 2015-03-30:
Hi,

before starting with some real application I am searching informations about getting and setting sensor/actor data right now. I read the tutorials, made some simple test programs and I noticed, that mmap fails for /sys/class/lego-sensor/sensor0/value0 and /sys/class/lego-sensor/sensor0/bin_data with errno == ENODEV.

So instead I use lseek to get to the first position and read one byte afterwards. This needs ~43µs. Usually I don't use mmap very much, so I wonder, whether this could be faster - especially as the file is no real file on the filesystem? In fact I don't even know, whether this value is particularly good or bad.

Or to ask more generic: What is the fastest userspace possibility to interact with the hardware?

Thanks!
Alex

dlech, 2015-03-30:
`mmmap` is not implemented for the `bin_data` attribute (although, I suppose it could be), and it can't be implemented for the `value*` attributes (they are not binary). However 3.3µs and even 40µs sounds reasonably fast, especially considering that most sensors only update at 100Hz (UART sensor may be a little faster). If we try to read the sensors at 1000Hz it uses about 10-15% of the CPU all of the time just for the analog/digital converter.

Or, to ask a question in return: why do you need to read sensors so fast?

theZiz, 2015-03-30:
Thanks for the fast answer! I didn't check the cpu load while reading the sensors. I did assume I use 100% for my tests. However, I don't need to check the sensor 300.000 times per minute. ;)
I just want to use as less cpu for the sensors/actors as needed, to have more cpu time for the P-stuff in the IPO Model. I want to make it right™ from the very beginning to not need to change it later. :)

dlech, 2015-03-30:
> I didn't check the cpu load while reading the sensors.

I actually meant the kernel drivers would be using 10-15% of the CPU all of the time, I wasn't even considering userspace. The kernel only reads the ADC at 100Hz, so even if you are reading `value0` or `bin_data` faster than that, you will just get the same value over and over until the ADC reads again (this statement only applies to analog sensors). The I2C operates at 10kHz, so considering that the minimum message size is 3 bytes = (8 bits + ack) * 3 + extra stop clock cycle = 28 bits => ~350Hz max read rate. Again, if you are reading the sysfs attributes faster than this, you will just get the same value in-between I2C messages.

> Another little thing I noticed

This is because they are different kinds of sysfs attributes. `bin_data` is a binary attribute. It is basically just a memcopy to get the data to userspace. `value0` is a "regular" sysfs attribute, meaning it is converted to a string. Additionally, `value0` is scaled, which can add some time. All of the code is [here](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/lego_sensor_class.c) if you want to check it out.

theZiz, 2015-03-30:
Thanks again for the fast (and at least for me: late) answer.
I will have a look at the code.
Furthermore I noticed, that my testing code was invalid. My speed testings did just test, how fast a system call can fail with wrong parameters... -_- So, it doesn't matter speedwise, whether you use the libc functions or a direct system call.

Thanks for all the clarifications. ;)

closed 2015-03-30 by theZiz