ev3dev/ev3dev #303: EV3 and Raspberry Pi via EV3 cable

HansAarneLiblik, 2015-03-31:
As some of you may know, basic EV3 can be connected to Arduino using basic sensor cable. One has to have a cable and an adapter for the Breadboard or a cable cut in half. Is this option available for Raspberry Pi and ev3dev for the to communicate, because this is a 2-way communication and then one could control sensors from Raspberry and send raspberry data to EV3

dlech, 2015-04-01:
This was discussed a bit in #253.

HansAarneLiblik, 2015-04-01:
There was a discussion about USB cable and local network. I'm talking about sensor cable for EV3 / NXT like they are doing in here
http://www.dexterindustries.com/howto/connecting-ev3-arduino/

dlech, 2015-04-01:
Yes, you could do that (communicate via I2C) with a Raspberry Pi. You could also do [this](https://lejosnews.wordpress.com/2014/06/03/arduino-uart-sensors/) (communicate via UART) with a Raspberry Pi.

So, if you are asking *can* it be done, sure, why not. If you want to know *should* it be done, I think it would be more practical/faster/easier to communicate via USB, Bluetooth or Wi-Fi.

HansAarneLiblik, 2015-04-01:
Communication via bluetooh or Wifi requires a dongle, Communication via USB (local network) requires a listener and a client on both ends. Since I2C is 2-way communication, I hoped it would be more easy to use for children

dlech, 2015-04-01:
I see what you mean. In that case, I would recommend emulating a UART sensor (like the link I already shared) with the Raspberry Pi rather than using I2C. If you use I2C, you will have to write a special driver for the EV3 or use low level I2C functions. If you emulate a UART sensor, then ev3dev will just detect it as a UART sensor and you can use the lego-sensor class to read the data.

closed 2016-02-25 by dlech