ev3dev/ev3dev #304: ev3dev home page

dlech, 2015-04-10:
It was brought up in #60 that our homepage at www.ev3dev.org needs some work. I've actually been meaning to do something about it for quite some time. So, here is my idea for a new page: <http://dlech.github.io/ev3dev.github.io/>.

Let me know what you think. It's still quite a lot of text, but ev3dev it a lot of things.

WasabiFan, 2015-04-10:
I like it! A few notes:

- Opening a web site to find the heading "What is is?" is generally a bad sign ;)
- The use of bullets as the main structure of the page adds both a lot of dead space and dots that seem out-of-place without any non-indented text around them. I'd recommend you try to break up those bullets in to paragraphs that provide the context needed, and use the bullets for one-sentence points. You might even be able to make those bullet points sub-headings.
- I like the idea of the boot screen image on the page, but I don't think it blends very well with the rest of the content (the background is a plain color that is different from the rest of the page, so it draws focus, but the image doesn't carry very much information). I had a similar idea back in #109 ([here specifically](https://github.com/ev3dev/ev3dev/issues/109#issuecomment-64459549)) that I think takes that idea and could make it seem more integrated with the rest of the page, although I think it could still use some work. Basically, I took those images of the terminal that @G33kDude posted and replaced the page header with them.
- We may want to consider an additional page for "non-geeks" that are just looking for information on the project without any geek-speak. I'm thinking of something that explains the effects of our different approach to the technology behind the OS without requiring you to understand what those technologies are. Then if someone wants to dig deeper (eg. learn what "Debian Linux" is), they can.

As an example of what I'd consider as a pretty good website for a project that is similar to ev3dev, you might want to refer to the website for [KnightOS](http://www.knightos.org/) for examples. They have content that is very similar in structure to what you made, and it is laid out in a very interesting and engaging way.

In the end, I think that your proposal is a much better design than what we have currently, but there are some little layout things that I think could make it even more approachable.

dlech, 2015-04-11:
> Opening a web site to find the heading "What is is?" is generally a bad sign

So what would you propose instead? (I noticed the KnightOS page has a "What is KnightOS?" section :wink:)

WasabiFan, 2015-04-11:
I'm not saying that the heading "What is it?" is a problem, but the title literally says "is is" -- there is an "s" instead of a "t".

dlech, 2015-04-11:
Got it. I've fixed that and the bullets. I'm thinking we save the "less geeky" page for when we get a little more polished and stable and it will be easier for less experienced programmers.

closed 2015-06-09 by dlech