ev3dev/ev3dev #306: Out of memory: Kill process nnn (brickman) score nnn or sacrifice child

BertLindeman, 2015-04-14:
Have seen this rather frequent on brickman v0.5.1 but also on v0.5.0.
Environment:
`Linux ev3dev 3.16.7-ckt6-ev3dev1 #2 PREEMPT Sun Feb 22 16:07:24 CST 2015 armv5tejl GNU/Linux`

I usually keep brickman at the main menu page, not at the file browser.

Part of syslog saved as [gist](https://gist.github.com/BertLindeman/b9cfc3ed5190db3a23e8)

My local *circumvention* is to restart brickman each night by cron

dlech, 2015-04-14:
Sounds like a memory leak. Just a hunch, but I am guessing that it has to do with Wi-Fi. Rather than restarting brickman, can you just disable Wi-Fi and see if it makes a difference?

BertLindeman, 2015-04-14:
Maybe a difference with other users? 
I have 3 accesspoints for wifi active and configured for the EV3.

The EV3 is on another floor here so I use wifi exclusively to connect to it.

I can disable wifi during the night and see if that makes a difference.
Would brickman notice if I use connmanctl to disable wifi via cron?
`connmanctl technologies disable wifi`
and enable again at daybreak.
Would it indicate the problem or is there a better way to do this?

dlech, 2015-04-14:
> Would brickman notice if I use connmanctl to disable wifi via cron?

Yes, brickman will notice. It is just the same as if you disabled wifi via brickman.

> Would it indicate the problem or is there a better way to do this?

If you check the memory usage of brickman and compare how much it grows while wifi is off compared to how much it grows while wifi is on, it will tell us if we have a memory leak related to wifi background scanning.

BertLindeman, 2015-04-14:
Thanks,
Will do so

BertLindeman, 2015-04-14:
First (probably too soon) result:
Changed 2 out of 3 accesspoints to NOT autoconnect.
brickman used near 30% mem according to `ps` and now - an hour later - %mem dropped via 
 27.7 and 25.7 to 25.3%
Taking ps measurements each 5 minutes.
Will **not** disable wifi for now, so I do not change multiple things at once.

BertLindeman, 2015-04-15:
Strange result.. (For me anyway)
brickman uses %mem between 25% and 35% from 22:00 until 06:00
And then drops in 5 minutes interval: 22.9% 17.4% 13.4% 10.9% 6.9%
After that mem usage climbs up to 38% from 07:00 to 18:00
Small graph:
[image]
Will leave the situation for another night and plan to switch off wifi at night tomorrow.

BertLindeman, 2015-04-17:
Looks like you are right @dlech.
Disabled wifi from 13:00 until 16:58 today.
Small graph of %mem taken from ps command:
[image]
A rather flat line while wifi was OFF and a climbing slope while wifi is ON.

dlech, 2015-04-17:
Good to know. Now I know where to start looking.

dlech, 2015-05-06:
I forgot to mention, this should be fixed in v0.6.0. Can you please test?

BertLindeman, 2015-05-06:
Have set OFF the connmanctl {en,dis}able wifi cron entries I had.
Last few days %mem reported by ps was lower than I was used to: now usualy around 6% with only two spikes to 14% but down after that on the next sample. Will keep a tap on it for a few days, but the figures are OK now, so closing.
Thank you @dlech

closed 2015-05-06 by BertLindeman

dlech, 2015-05-06:
\o/