ev3dev/ev3dev #307: Common lisp on ev3dev

peebz, 2015-04-15:
Hi everyone,

Has anybody been successful in installing common lisp on ev3dev?  I've tried several that have ARM builds (SBCL, CCL, and ECL) but all seem to have problems running.  I'd be very grateful for any suggestions or advice.

Cheers,

David

theZiz, 2015-04-15:
How did you try to install them? I don't have my Ev3 here atm and I never used lips, but ecl seems to run fine within my brickstrap qemu session. I installed it via
sudo apt-get install ecl
I tried a (googled) hello world and it seems to work there. What doesn't work for you?

Greetings,
Alex

peebz, 2015-04-15:
Many thanks for your rapid response Alex.

I installed ECL via apt-get (the other two by downloading and manually installing).  This issue I was having with ECL was when I tried to compile a file (which didn't occur when I did the same on another non-ARM Debian system.  I seem to have resolved the issue - I think by installing libffi-dev.

Thanks again,

David

dlech, 2015-05-21:
So this is working and we can close the issue?

closed 2016-02-25 by dlech