ev3dev/ev3dev #308: servo-motor class is not working for a hobby servo

erichuang2013, 2015-04-16:
I am trying to control a normal RC ESC for servo-motor class. but don't have a clue how to generate the servo-motor-class under /sys/class. any hardware wiring needed to allow ev3dev to detect external device? or I can manually add this servo-motor class in a specific port?

My kernel version is Linux "ev3dev 3.16.7-ckt6-ev3dev1 #2 PREEMPT Sun Feb 22 16:07:24 CST 2015"

dlech, 2015-04-16:
Currently, the only supported servo controller is the [mindsensors.com 8 channel servo controller](@erichuang2013, currently, the only supported servo controller is the mindsensors.com 8 channel servo controller. You need one of these to use your servo.). You need one of these to use your servo.

dlech, 2015-04-16:
The output ports on the EV3 use battery voltage (up to 9V), which is too much for most servo motors, so it is not possible to connect the servo directly to the EV3 without some sort of electronic buffer and writing a new driver.

dlech, 2015-12-12:
As I already stated, based on my understanding, I don't thing this type of servo should be directly connected to the output ports on the EV3, so closing this issue.

closed 2015-12-12 by dlech