ev3dev/ev3dev #309: Suggestions on improving startup sequence

sharived, 2015-04-16:
Here are my instructions for setting up Bluetooth on [Mac](https://github.com/10Imaging/RoboSnap/wiki/Ev3-on-Mac) and [Windows](https://github.com/10Imaging/RoboSnap/wiki/Ev3-on-windows).  It still seems rather complicated for kids to use.  In particular the fact that I need to plug in a keyboard for the Windows passkey.  Do you have any suggestions on improvements I can make to this process?  At one point I thought about replacing brickman with my own UI - but haven't attempted that.  Appreciate any feedback.

dlech, 2015-04-16:
I think you will have better luck using tethering. The main reason for saying this is there have been a number of people who have reported problems establishing a bluetooth network connection with Windows. It works fine when using the Microsoft Bluetooth stack that comes with Windows, but if it uses a 3rd party Bluetooth driver stack, then networking may not be available.

When tethering is enabled on the EV3, it provides the Bluetooth network connection. To the PC, it looks pretty much like a smartphone with tethering enabled, which is well supported. Also, with tethering, you don't have to mess with any of the tabs on the networking screen.

---

You can enter the passkey on the EV3 by using the arrow keys. You just have to press the enter button first to enable the cursor.

---

It may be simpler to initiate pairing from the desktop computer instead of the EV3. It works the same either way.

---

Beyond that, you could write a program that runs on Mac and Windows that takes care of the Bluetooth stuff automatically instead of having to manually configure it. This should be easy to do on Mac, but not sure about Windows since there is the possibility of different Bluetooth stacks.

sharived, 2015-04-16:
Thanks.  Tethering doesn't seem to work for my application - I am communicating to the robot via http.  Thanks for the hint about the keyboard - that simplifies things.

I do initiate pairing from the windows machine, I'll try from the Mac as well.

Is there a way to get "visible" to stay on.  I seem to have to set it each time.  That would simplify the process a bit.

dlech, 2015-04-17:
Tethering sets up a network connection, so I don't see why it wouldn't work - unless the http server is not binding to the `tether` interface for some reason.

for the "visible" issue, I have uncommented the line

    DiscoverableTimeout = 0

in `/etc/bluetooth/main.conf` which keeps it visible indefinitely. Although, if you initiate the pairing from the EV3, then the EV3 does not need to be discoverable.

sharived, 2015-04-29:
That worked thanks!  Closing issue.

closed 2015-04-29 by sharived