ev3dev/ev3dev #310: License

HansAarneLiblik, 2015-04-17:
I'm doing my bachelor thesis on EV3 and RPi and using your firmware. I really could use some images provided at ev3dev.org and some other material in docs.

Question is, is this project licensed under Creative Commons? I could not find any license information online about this project. 

Thank you!

dlech, 2015-04-17:
This is a great question and something we probably should have thought about a long time ago. Last night, I was searching the web for "ev3dev" just to see what people are using it for. I found several instances were people have copied our documentation without asking and without giving credit. I don't mind that they copied our stuff, but giving credit when you do so is just common courtesy. So, thank you for asking! :+1: 

So, in the spirit of open source, I am proposing that we license the www.ev3dev.org website using the [Creative Commons Attribution-ShareAlike 4.0](http://creativecommons.org/licenses/by-sa/4.0/) license.

To save some trouble, I think we should omit the projects pages from this and let it be up to the project authors as to how they would like to share their works.

So, with the permission of @rhempel and @WasabiFan, who have also contributed significant portions of the website, I will add a copyright/license statement to the website.

And since you asked about the project as a whole, it also includes lots of source code. All of the source already has its own copyright/license information (mostly GPL2).

rhempel, 2015-04-17:
Fully agree - Creative Commons Share Alike is fine. Many people don't bother reading Licenses, so if we can add a clear statement that if someone copies anything from the site that the very least they can do is give us credit and link back to ev3dev.org

HansAarneLiblik, 2015-04-17:
Thank you ! It is helping me a TON!

WasabiFan, 2015-04-17:
> Fully agree - Creative Commons Share Alike is fine.

Me too! CC w/ attribution sounds good.

dlech, 2015-04-17:
Done!

https://github.com/ev3dev/ev3dev.github.io/commit/a4ba3f3871e75d8722bec199ea575ddde3acbd4c

closed 2015-04-17 by dlech

dlech, 2015-04-17:
> Thank you ! It is helping me a TON!

You are welcome. And we would love to see your work when your are finished. :wink: