ev3dev/ev3dev #311: Sensor docs describe a "name" attribute?

WasabiFan, 2015-04-18:
On [this line](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/lego_sensor_class.c#L41) of the sensor docs definition, a "name" attribute is referenced (in a code block), but I don't see that property listed in the docs. Is this a remnant of some old text, or is there something missing? If this is intended to refer to the `driver_name` and `port_name` properties, I think that that "`name`" should be changed to "`*_name`" to make it clear that you can use either to identify the device.

dlech, 2015-04-19:
Yes, that was leftover from a previous version. The correct attribute is `driver_name`.

WasabiFan, 2015-04-19:
Thanks for clarifying!

closed 2015-04-19 by WasabiFan