ev3dev/ev3dev #312: Ev3Dev on other platforms?

sharived, 2015-04-18:
As I'm trying to work with some other platforms, I'm really preferring your motor controls as opposed to what is already existing.   Have you ever thought about getting Ev3Dev to work with platforms such as RaspberryPi+BrickPi or the Arduino ones (I'm waiting for the new kickstarter shield) and there was also a BeagleBone Black cape that just came out.  Would be nice to have a standard interface for all these "Ev3" platforms.
Anyway, just a thought.  I'm sure there is plenty of work just supporting one hardware platform.

dlech, 2015-04-19:
Send me the hardware and I will make it work :wink:

Seriously, I would love to see this happen. This thought has been in the back of my head since I have seen the BrickPi and the EVB. We have designed the drivers and brickman with this possibility in mind. So, yes, it is plenty of work supporting one platform, but I don't think it would be too much more work to support multiple platforms. I don't have the :moneybag: laying around to invest in all of these hardware platforms just so I can hack on them, but I am willing to trade my expertise for some fun toys to play with that I otherwise wouldn't allow my self to have.

Arduino stuff won't work with ev3dev though because it doesn't run Linux.

sharived, 2015-04-19:
awesome.  I will work on getting you the hardware.  Stay tuned!

rhempel, 2015-04-19:
I'll second @dlech on that, Shari - please send any hardware (or financial donations) to David so that he can continue to do what he does so well, which is make it a bit easier for anyone to get into robotics.
When I originally started the ev3dev project, the goal was to buld up a platform for others to use - the decision to share it publicly and open the development up was a direct result of my earlier work with pbLua (a Lua for MINDSTORMS NXT). I was always under the gun to get things working because I didn't want anyone else to help.
I'm very happy that resources like GitHub make the kind of collaboration we have on the ev3dev team possible, and that @dlech has stepped in an become one of the driving forces behind ev3dev.

DexterInd, 2015-04-20:
Hey folks, we're happy to send a BrickPi to @dlech a BrickPi.  I just shot out a message through your website.  If that didn't work, send me a message through here:  http://dexterindustries.com/contact.html

moriarty, 2015-04-20:
:+1:  @DexterInd 

I have a BrickPi from the kickstarter campaign - but have had very little time to use it, and then got my hands on an ev3. If @dlech gets his hands on one I will try to find some time to help.

rhempel, 2015-04-20:
Thanks for the support - @dlech is definitely the guy that can get it to work! He has made significant contributions to my initial proof of concept and helped to turn it into a serious platform for LEGO EV3 development. 
By adding support for other hardware platforms that can conform to the relatively simple device control model that we developed - the goal will be to be able to run very similar programs on the EV3 and the rPi family of modules.
The additional RAM on the rPi is a definite plus....

sharived, 2015-04-20:
@dlech - I'll send the Raspberry Pi - where should I send it?

sharived, 2015-04-21:
I have the address now - will ship tomorrow.

dlech, 2015-04-21:
Sneaky...

rhempel, 2015-04-21:
Long story @dlech - the customs stickers on the envelopes were swapped - somehow the USPS decided to use the address on the front of one envelope and the sticker on the back of the other.
It's like one of those probablility questions in high school :-)
Look on the bright side, you'll get custom LEGO tiles and an rPi on the same day!

dlech, 2015-04-23:
Just got the package from @sharived / @rhempel!

Thanks guys!

JorgePe, 2015-05-03:
Hi!
It's time for me to also donate. I read you would like to have a Pi 2, so you can count with one from me. Please send me the address details.

DexterInd, 2015-05-03:
Did our BrickPi donation make it?

dlech, 2015-05-03:
> Did our BrickPi donation make it?

Yes, I received it on Friday. Thank you very much!

If you would like to follow along with the development, just watch any of the repositories in the ev3dev organization on GitHub that have `rpi` in the name.

So far, I have it booting to jessie and I have created a device tree overlay that enables I2C and sets up `leds` devices for the LEDs on the BrickPi.

DexterInd, 2015-05-04:
Good to hear!  We will follow along!

MetallianFR68, 2015-05-04:
Thanks Dexter Industries for that donation.
I'm very interested in the Brickpi and I wonder what dlech is going to achieve with it.

Is there a way to stack 2 or 3 brickpi on the same RPI ?

DexterInd, 2015-05-04:
Our pleasure!  We can't wait to see what @dlech does as well!

We haven't developed a way to stack 2 or 3 brickpi yet.  The protocol between shields is in Serial; it's possible, but it would just require more work on the firmware.

MolarAmbiguity, 2015-05-04:
@DexterInd have you made any progress on supporting ev3 sensors? Last I read, brickpi wasn't working with them?

DexterInd, 2015-05-04:
@MolarAmbiguity, indeed, we support EV3 sensors.  It required a new firmware, which is shipping with all of our units now.  

Examples are here: https://github.com/DexterInd/BrickPi_Python/tree/master/Sensor_Examples

MolarAmbiguity, 2015-05-05:
@DexterInd Cool you may see an order for a brickpi from me soon. ;)

ameysutavani, 2015-05-07:
@dlech @DexterInd ...This is awesome collaborated development.

dlech, 2015-05-13:
It's here!

http://www.ev3dev.org/news/2015/05/13/raspberry-pi-image-release/

sharived, 2015-05-13:
Awesome!  Will give you some feedback next week.  Busy with Maker Faire
coming up this weekedn and launched my kickstarter today
https://www.kickstarter.com/projects/robosnap/robosnap-vision-for-your-robot
(Working on a project page for it -let me know if there is anywhere else I
can post to Ev3Dev fans!)

dlech, 2015-05-13:
The Kickstarter page looks great. Best of luck to you. I'll add a post about it to our news feed.

dlech, 2015-05-13:
Oh, and there are quite a few LEGO MINDSTORMS educators at https://www.facebook.com/groups/legomindstorms/ that would be interested.

dlech, 2015-05-20:
Going to go ahead and close this one. I'll open a new issue for EVB support . If there are problems or suggestions for the BrickPi drivers, those should go in new issues as well.

closed 2015-05-20 by dlech