ev3dev/ev3dev #313: Reducing brick power consumption

prabindh, 2015-04-19:
When working off batteries, it is needed to optimise the power consumption.  I have below ideas, and need some discussion around it:

- Switch off LEDs (currently all LEDs are on and triggered)
- Reduce CPU speeds (more aggressive governors, currently only CPU_IDLE is enabled)
- Switch off display (default is ON today)
- Switch off connectivity (default is OFF)

I submitted a patch for the LEDs as below. Any thoughts on the other items ?
https://github.com/ev3dev/ev3dev-kernel/pull/6

dlech, 2015-04-19:
If you want to squeeze every last drop of power out of the EV3, then I suppose these might make some difference. Do you have a current meter where you could actually measure how much power you are saving by making these changes? I have played around with some of these in the past, and the difference is in most cases is negligible.

> Switch off LEDs (currently all LEDs are on and triggered)

The LEDs probably make the biggest difference. As I commented in your pull request, you proposed changes are changing something that should not be changed and will not actually do what you are suggesting. I don't really think powering off the LEDs for the 60 seconds it takes to boot will save that much power. If you want to turn off the LEDs after boot is complete, then you should copy `/lib/systemd/system/ev3dev-leds.service` to `/etc/systemd/system/ev3dev-leds.service` and change it accordingly.

> Reduce CPU speeds (more aggressive governors, currently only CPU_IDLE is enabled)

I don't really have any experience in this area, so if you or anyone else does, I am open to suggestions. Although I will say that I have never noticed any difference in current consumption with 100% cpu usage vs. low cpu usage.

> Switch off display (default is ON today)

Since the display is an LCD with no backlight, it uses on the order of microamps to show something on the display. The biggest power consumption of the display is when it is updated. As long as nothing on the display is changing, nothing is updated. I have written bricman to change the display as little as possible to make sure we are not wasting power. But if you really want to, you can disable brickman and tty logins, then the display will use virtually no power.

> Switch off connectivity (default is OFF)

Bluetooth is really the only built-in device that makes a difference here, and yes, the radio is turned off by default.

prabindh, 2015-05-17:
@dlech  I am getting a 24% reduction in current consumption with the patch. Detailed numbers for the power consumption on the LED and the overall system is in the link below. Please check and confirm.

http://robotpowered.org/ev3power.txt

rhempel, 2015-05-17:
I think the actual current drop is more like:

```
110 mA - Current when all 4 LEDs (left/right, RED/GREEN) are on with default level of brightness (255) - 110 to 112 mA
5 mA   - Average drop of current, when level of brightness of LED is reduced from 255 to 64 or 0 - 5-6 mA
```

5/110 which is more like 4% - you need to measure the LED on vs LED off current during idle operation so that the difference makes any sense. I would have expected those LEDs to be drawing about 1-2 mA each at full power, so the 5-6 mA drop when they are off makes sense.

prabindh, 2015-05-17:
The 5mA is per LED ....

dlech, 2015-05-17:
I have repeated your experiment and got similar results. I was actually kind of surprised the the current consumptions was so low at idle. Either I must usually run at a lower voltage or have more things plugged in. On average, I would say that my EV3 usually runs between 150mA and 200mA.

I ran my experiment with a power supply at 8V. Nothing is plugged in to the EV3 except for USB to the host computer. Bluetooth is off (which actually does not seem to make a difference if it is on and not scanning).

* During boot, I saw fluctuations between about 95 and 110mA.
* During idle with both green LEDs on (100%), I have 96mA.
* During idle with all LEDs off, I have 82mA.
* During idle with both red LEDs (100%) on, I have 92mA.

So, I get 7mA/56mW per green LED and 4mA/32mW per red LED.

Definitely a potential for some significant power savings here. As I mentioned in ev3dev/ev3dev-kernel#6, I think the best way to handle this would be to make the `max_brightness` of the LEDs user configurable via a kernel command line option. This way, the max brightness will take effect for all userland programs (assuming they are written the "right way") instead of having to modify all userland programs as well.

rhempel, 2015-05-17:
Good work! I'm a little surprised at how much current those LEDs draw to be honest. I support the user configurable `mx_brightness` idea too - much better that we touch one thing and all apps work the same.

ensonic, 2015-07-01:
Regarding cpu-governors, not sure how much the arm9 can do. But for a start it seems that the related kernel modules are not built:

    ls -1 /lib/modules/`uname -r`/kernel/drivers/cpufreq/

There seems to be a driver for davinci:
http://lxr.free-electrons.com/source/drivers/cpufreq/davinci-cpufreq.c
For some links:
https://wiki.archlinux.org/index.php/CPU_frequency_scaling

ensonic, 2015-07-01:
FYI: powertop runs on the ev3 albeit super slowly, so if you try, be patient! It nly has two tweakables it suggests. One does not make a meassurable different (vm_write_back) and I yet have to test the suggested powersaving for wifi-usb.

dlech, 2015-07-01:
If someone wants to try playing around with the cpufreq kernel modules and let us know the results, we would love to learn more about it. I certainly don't have any expertise in that area.

I tried running powertop early on in working with ev3dev, but I couldn't get it to run (I think that was before jessie). Good to know that it does work. USB Wi-Fi power savings would definitely be a plus.