ev3dev/ev3dev #315: Print bot code?

mmattyg, 2015-04-24:
Hi, not really an issue, I'm looking for the code that operates @cavenel 's EV3 Print bot (the one over here https://www.youtube.com/watch?v=UwpghsdmDsE)

I built my own model, and wrote some code, but having a lot of difficulties with the math, and can't get it to draw a simple square (I get a trapezoid in the best case). Couldn't find the code anywhere, It will really be helpful if someone (@cavenel himself, if you're reading here?) can post the code

Thanks! Appreciate the help
-Matty

cavenel, 2015-04-24:
Hi,
Unfortunately, the code is not up to date as I didn't have the time lately to play with ev3. And as ev3dev changed a lot of parameters names lately, it will not work at all.
I can still send you my code for information, I will try to find some time to clean /comment it a bit and will put it on my github this week-end.
All code is in Python.

mmattyg, 2015-04-24:
thanks @cavenel 
I'm actually using a NXT brick and  Roboc, so I don't need a code that works 'out of the box'
if you can just point out what the model-specific variables are called (the arm lengths, gear ratios etc) so that I can use the math that converts the motor angles to a pen XY,  that would be very helpful
-Matty

mmattyg, 2015-04-27:
@cavenel  nagging you again.. didn't mean to make extra work for you. Just whatever you got there is fine as-is, and will be much appreciated..
thanks (sorry)
-Matty

cavenel, 2015-04-27:
Sorry, I did not have time to look at that this week-end. Here is some (python) code I used to convert motor position to coordinates and the other way around, maybe that can already help you?
The code to draw things is a bit more complicated as I had to bypass a bug with rotate_forever and regulation. But I can look at that during the week.
Christophe

    ## Coordinates of origin of arm A (in Lego measures)
    xA, yA = 0.,0.
    ## Coordinates of origin of arm B (in Lego measures)
    xB, yB = 6.,0.
    ## r1 length of first part of arms AC and BD (in Lego measures)
    r1 = 15.+1.3125
    ## r2 length of second part of arms CE and DE (in Lego measures)
    r2 = 11.

    #      .E   (pen is incoordinates E = (xE,yE))
    #     / \
    #    /   \
    #   /     \
    # C.       .D
    #   \     /
    #    \   /
    #    A. .B
    #   -------
    #   [robot]
    #   -------

    ## Computes the intersection of 2 circles of centres x0,y0 and x1,y1 and radius resp. R0 and R1.
    def get_coord_intersec (x0, y0, x1, y1, R0, R1):
        if y0 == y1:
            y0+=0.01
        N = R1*R1 - R0*R0 - x1*x1 + x0*x0 - y1*y1 + y0*y0
        N /= 2*(y0-y1)
        A = ((x0-x1)/(y0-y1))*((x0-x1)/(y0-y1)) + 1
        B = 2*y0 * (x0-x1)/(y0-y1) - 2*N*(x0-x1)/(y0-y1) - 2*x0
        C = x0*x0 + y0*y0 + N*N -  R0*R0 - 2*y0*N
        delta = math.sqrt(B*B - 4*A*C)
        xA = (-B + delta) / (2*A)
        xB = (-B - delta) / (2*A)
        yA = N - xA * (x0-x1)/(y0-y1)
        yB = N - xB * (x0-x1)/(y0-y1)
        return (xA,yA),(xB,yB)

    ## Converts coordinates xE, yE to angles of robot arms.
    def coordinates_to_angles (xE, yE):
        try:
            ((xC1, yC1), (xC2, yC2)) = get_coord_intersec (xE, yE, xA, yA, r1, r2)
            if xC1 > xC2:
                xC1 = xC2
                yC1 = yC2
            ((xD1, yD1), (xD2, yC2)) = get_coord_intersec (xE, yE, xB, yB, r1, r2)
            if xD1 < xD2:
                xD1 = xD2
                yD1 = yD2
        except:
            return None  
        alpha = 180. - 360 * math.acos((xC1-xA)/r2) / (2.*math.pi)
        beta =  360 * math.acos((xD1-xB)/r2) / (2.*math.pi) 
        return (alpha, beta)

    ## converts coordinates x,y into motor position
    def coordinates_to_motorpos (x, y):
            #14    => 0
            #90    => -3158
        def angle_to_motorpos (angle):
            return int((angle-14) * 3158 / (90-14))
        (alpha, beta) = coordinates_to_angles (x, y)
        return angle_to_motorpos (alpha), -angle_to_motorpos (beta)
        
    ## Converts angles of arms to coordinates.
    def angles_to_coordinates (alpha, beta):
        xC = xA - r2 * math.cos((2.*math.pi) * alpha/360.)
        yC = yA + r2 * math.sin((2.*math.pi) * alpha/360.)
        xD = xB + r2 * math.cos((2.*math.pi) * beta/360.)
        yD = yB + r2 * math.sin((2.*math.pi) * beta/360.)
        ((xE, yE), (xE2, yE2)) = get_coord_intersec (xC, yC, xD, yD, r1, r1)
        if yE2 > yE:
            xE = xE2
            yE = yE2
        return xE, yE

    ## Converts motor position to coordinates
    def motorpos_to_coordinates (pos1, pos2):
        def motorpos_to_angle (pos):
            #0     => 14
            #-3158 => 90
            return 14 + pos * (90-14) / 3158.
        
        (alpha, beta) = (motorpos_to_angle(pos1), motorpos_to_angle(-pos2))
        return angles_to_coordinates (alpha, beta)

rhempel, 2015-04-28:
Christophe - please send me your mailing address and I will send you
some ev3dev.org tiles to decorate your creations :-)

Ralph

mmattyg, 2015-04-28:
Thanks @cavenel ! Just what I needed
Can I ask how you calculated these numbers (14, 3158)? (My Lego model is sized a bit different)
```
def angle_to_motorpos (angle):
        return int((angle-14) * 3158 / (90-14))
```

Also, the "beta" angel is negative, because the right motor spins the gear to the other direction, right?
```
return angle_to_motorpos (alpha), -angle_to_motorpos (beta)
```

cavenel, 2015-04-28:
Exactly for the negative beta, each motor turns in opposite directions if I
remember correctly.
For 14 and 3158, I got these values from empirical measures. Put first arm
at angle minimum (I measured 14 degrees for me) and initialize motor
position to 0. Rotate motor until arm is at 90 degrees and look at new
motor position (3158 for me). As both arms should be exactly symmetric, you
know that the other motor will go from 0 to -3158 for the same angles.
This will tell you the correspondence between angle and motor position, as
it should be linear between the min and 90 deg.

For information, when I want to go to a coordinates (x,y), I look at actual
coordinates from motor positions, and compute the vector from actual
coordinates to x,y. Then compute coordinates that are in direction of this
vector but at a distance of 1 cm or less (to have the "local displacement"
to the target). Then I compute the motor positions for this new point and I
fix speed of both motors proportionally to go there. And I repeat this
every 0.1 sec until destination is reached.
This is maybe not very clear but it's the only way I found if you want
straight lines (going directly to final point will lead to roundish curves
as each motor will have constant speed)
If you want more info on this part I can give you some code tomorrow that
could make it clearer. I don't have access to it right now.

mmattyg, 2015-04-29:
Thank you, that make a lot of sense. And sure, I'd love to see the other part of the code

cavenel, 2015-04-29:
Glad if it makes sense :) Here is a simplified version of the function to reach a point x,y:

    def goto_point (x,y):
        # Compute the motor positions of destination x,y
        final_pos1, final_pos2 = coordinates_to_motorpos (x, y)
        close_to_dest = False # We loop until we are close enough to destination :
        while (not close_to_dest):
            # Get actual coordinates as myx, myy, and distance to destination x,y as dist.
            myx, myy = motorpos_to_coordinates (mot_droite.get_position(),  mot_gauche.get_position())
            dist = math.sqrt((myx-x)*(myx-x) + (myy-y)*(myy-y))

            # Check if we are close to destination (closer than 0.1 lego units)
            if dist > 0.2:
                # If not close, get point in direction of destination and at distance 0.2 lego units.
                nextx = myx + (x - myx) / (dist * 5.)
                nexty = myy + (y - myy) / (dist * 5.)
                next_pos1, next_pos2 = coordinates_to_motorpos (nextx, nexty)
            else:
                # If close, our next destination is the final destination
                next_pos1, next_pos2 = final_pos1, final_pos2
                close_to_dest = True

            # next_pos1, next_pos2 contain the resp. positions of motors we aim for.
            
            # Adapt maximum motor speed to distance to destination, so that drawing slows down in the end
            speed_max = (40 if dist > 2 else 20 + dist * 10)
                
            # Compute distance for each motor to reach his  destination
            dist1 = (next_pos1 - mot_droite.get_position())
            dist2 = (next_pos2 - mot_gauche.get_position())
            
            # Compute speed for both motors with speed max as comuted above (=speed_max).
            if math.fabs(dist1) > math.fabs(dist2):
                speed1 = speed_max * dist1 / math.fabs(dist1)
                try:
                    speed2 = math.fabs(speed1) / math.fabs(dist1) * dist2
                except:
                    speed2 = 0
            else:
                speed2 = speed_max * dist2 / math.fabs(dist2)
                try:
                    speed1 = math.fabs(speed2) / math.fabs(dist2) * dist1
                except:
                    speed1 = 0

            # Give each motor his own speed, with aimed position as original final destination (so that even if lag, final destination will be reached)
            mot_droite.goto_position(final_pos1, speed1, regulate=1, brake=False)
            mot_gauche.goto_position(final_pos2, speed2, regulate=1, brake=False)
            time.sleep(0.01)

mmattyg, 2015-04-29:
I now see that I was going about this the wrong way.
I'll keep you posted, thanks for the help.  I hope to post a video of a straight line soon..

mmattyg, 2015-05-02:
btw in ```coordinates_to_angles``` where is says
```((xD1, yD1), (xD2, yC2)) = get_coord_intersec (xE, yE, xB, yB, r1, r2)```
it should probably be (xD2, yD2) [instead of yC2)

cavenel, 2015-05-02:
You are right, I changed some variable names to make it easier to understand, must have failed my search/replace ;)

closed 2016-02-25 by dlech