ev3dev/ev3dev #316: Create ev3dev-config tool

dlech, 2015-04-29:
As discussed in #280, we think it would be nice to have a simple tool to do most of the [first things](http://www.ev3dev.org/docs/getting-started/#step-5-first-things-to-do-with-ev3dev) from the ev3dev getting started page. Raspian (Debian wheezy on Raspberry Pi) already has a tool called `raspi-config` to do this that I think we could build from.

If anyone has been looking for something they could do to contribute to ev3dev, here is a good opportunity. This should be a fairly easy task if you are not afraid of shell scripts. It's mostly removing stuff anyway.

I have cloned [raspi-config](https://github.com/asb/raspi-config) into [ev3dev-config](https://github.com/ev3dev/ev3dev-config), so fork [ev3dev-config](https://github.com/ev3dev/ev3dev-config) and get hacking.

Here are the changes I think need to be made:

- [ ] Add a `README.md` file explaining what this is. Be sure to include a note that this is a fork of `raspi-config`.
- [ ] Rename files and replaces all instances of `raspi-config` with `ev3dev-config`.
- [ ] Remove the "Expand Filesystem" bits. This will be included in the firstboot script.
- [ ] Remove the "Enable Boot to Desktop/Scratch", "Enable Camera", "Add to Rastrack", "Overclock" and "Advanced Options" menu items. Can leave the functions in case we decide we need them later.
- [ ] Modify the `do_change_hostname()` to use `hostnamectl` instead.

MolarAmbiguity, 2015-05-02:
So is the idea to have ev3dev-config at first boot, then use the brickman menu for any further configuration? Or will ev3dev config be used for any configuration that needs to happen?

dlech, 2015-05-03:
`ev3dev-config` is for the stuff that you don't do very often, so you forget how to do it - and for stuff that would not work well on the small screen or without a keyboard.

brickman if for things you might want (or need) to do without having a separate computer connected.

WasabiFan, 2015-12-30:
I _would_ volunteer for this now, but I am pretty well-loaded with ev3dev projects at the moment. If someone doesn't claim this within a few weeks, I might be able to get this started.

WasabiFan, 2016-02-25:
Cool, I'm officially assigned to something! :grin:

dlech, 2016-02-25:
That's what happens when you make significant contributions. :wink:

dlech, 2016-02-25:
Hmm... I'm wondering if we should just include this in the ev3dev-tools package instead of a separate repository instead of having a separate repository. We already have lots of `*-config` packages, but the others all have to do with config files (as in `/etc`).

WasabiFan, 2016-02-25:
I have no vote either way... All I'll say is that the sysinfo tool is pretty lonely at the moment in its repo :wink:

dlech, 2016-02-25:
Do you want to copy and paste the `ev3dev-config` script to `source/scripts` in [ev3dev-tools](https://github.com/ev3dev/ev3dev-tools) and make a pull request? Name the new file `ev3dev-config.sh`.

WasabiFan, 2016-02-25:
Give me a minute and I should have a PR ready.

dlech, 2016-02-25:
This has been released in `ev3dev-toos 1.1.0`.

closed 2016-02-25 by dlech