ev3dev/ev3dev #317: Turning off sensor

donwojtallo, 2015-05-03:
Is there any way to turn of sensor? I can't find that on docs. 
I'm exactly trying to disable color sensor during the program execution so the light will be temporary turned off.

dlech, 2015-05-03:
I assume you mean the [LEGO EV3 Color Sensor](http://www.ev3dev.org/docs/sensors/lego-ev3-color-sensor/). The "Description" column in the "Modes" table tells you the LED state for each mode, so to turn the LED off you need to put the sensor in `COL-AMBIENT` mode.

donwojtallo, 2015-05-03:
This does not turn off LED, ```COL-AMBIENT``` changes its color to blue. And other modes switches color between red, blue and white. The only mode that turns off the light is ```COL-CAL```, but then the LED starts flashing and goes back to red after few seconds.
Anyway, I was wandering if there is any way to disable completely the functionality of sensor so the brick could save some power...

dlech, 2015-05-03:
You are right. It looks like we need to fix the documentation. (Could I talk you into doing that for us? There is an "Edit on GitHub" link on that page.)

Instead, you can change the mode of the port the sensor is connected to using the `lego-port` class. Find the port in `/sys/class/lego-port/` with the matching `port_name` attribute. Then use the `mode` attribute of the port to change the mode. The least intrusive mode for "disabling" the sensor is probably the `other-uart` mode. Then, you can switch back to `ev3-uart` (or `auto`) when you need to use the sensor again. You also need to know that when you do this, the `lego-sensor` device will be removed, so when you enable the sensor again, it will show up as a new device.

donwojtallo, 2015-05-04:
Great, it works exactly as you described. Thanks ^_^.
I also created a pull request with fixed docs.

closed 2015-05-04 by donwojtallo