ev3dev/ev3dev #318: (null) on motors list (Brickman v0.6.0)

donwojtallo, 2015-05-03:
I revently updated ev3dev by doing:
```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo reboot
```
Now in Brickman v0.6.0 on motor list i see ```(null) on outA``` (or other port), no matter to which port and motor i plug in. I also cannot enter the submenu of the motor. When I hit enter while the motor is highlighted, nothing happens.

I tried to update firmware 2 times more, also tested 3 motors and all outputs, the menu always shows ```(null)```.

dlech, 2015-05-03:
Brickman v0.6.0 is designed for use with kernel version 3.16.7-ckt10-4-ev3dev-ev3. However, 3.16.7-ckt10-4-ev3dev-ev3 will break existing installations, so it does not upgrade automatically. Please read the full [announcement](http://www.ev3dev.org/news/2015/05/01/Major-Release/) about the latest release.

donwojtallo, 2015-05-04:
Thank You! Now its shows correct device name.

closed 2015-05-04 by donwojtallo