ev3dev/ev3dev #319: tacho-motor/motorX/type missing

GusCaplan, 2015-05-05:
Hello,

I was using attempting to use the ev3dev nodejs and when running the motor example I got an error that the /type was missing from the tacho-motors and indeed it was for all motors big and small. It just isn't there. I have tried rebooting.

dlech, 2015-05-05:
We need to know which kernel version you are running (please see <http://www.ev3dev.org/support/>).

`tacho-motor/motorX/type` was removed several versions ago and the nodejs library has probably not been updated yet.

WasabiFan, 2015-05-05:
@GusCaplan Are you using the version from npm or the version from GitHub? The version from npm is not up-to-date for the latest kernel. If you are looking to use it with the latest kernels, you can download and build it from [the JS repo](https://github.com/wasabifan/ev3dev-lang-js). If you don't want to build it yourself, I might be able to release an up-to-date testing release to npm in the next few hours.

GusCaplan, 2015-05-05:
@dlech I am running whatever kernel comes with the ev3dev-jessie-2015-05-01 build. (I can't run uname right now sorry).

@WasabiFan if teh git repo will support the /type-less tacho-motors then I will build from there

GusCaplan, 2015-05-05:
@WasabiFan an official build wouldn't go amiss either.

GusCaplan, 2015-05-05:
@dlech its the 3.16.7-ckt10-4-ev3dev-ev3 kernel

WasabiFan, 2015-05-19:
@GusCaplan I just released an up-to-date version of the Node binding [here](https://www.npmjs.com/package/ev3dev). I'd love some feedback on how it is!

P.S. If this solves the problem, can you close the issue?

GusCaplan, 2015-05-20:
@WasabiFan it fixes it.

closed 2015-05-20 by GusCaplan