ev3dev/ev3dev #320: Changing position of running / holding motors

cavenel, 2015-05-06:
I updated to ev3dev-jessie-2015-05-01 recently and have been changing my codes to run with new tacho-motor class. For calibration of motors, I sometimes fix position = 0 after moving the motor. This was working without any problem before in all cases. 
With the new API, I sometimes receive a message about "Device or resource busy". I see that it is the expected behavior if motor is running (here : http://www.ev3dev.org/docs/tutorials/tacho-motors/#position-and-speed)
Unfortunately, it is the case if motor is holding (stopped) as well or if the motor is stopped recently in coast mode and still moves. It makes it tricky to change position at a specific point (rotating motor to a specific position, then keep position with hold and reset position) and forces to wait a bit when motor just stopped.
My questions are:
* Is this busy bit important and / or useful ? It was very practical to change position at any time even on running motor.
* If it is important, what is the best / fastest way to reset a motor position that just stopped? Is it enough to check motor state before and then change position?

Details on my config:
* Image file: ev3dev-jessie-2015-05-01
* Kernel: v3.16.7-ckt10-4-ev3dev-ev3
* Update method: from image file (not via apt-get upgrade)
* Not dependent on library or binding.

dlech, 2015-05-06:
> Is this busy bit important and / or useful ?

Yes, there are a number of reasons for this.

1. If the command is `run-to-*-pos` and the position was changed before the position was reached, the direction of the motor may have to be changed. This should never happen without sending a new run command.
2. When stopped and `stop-mode` is `hold`, the controller is using a PID with `position` as the process variable. If you change the process variable by a large amount, it will upset the PID and the motor jerk suddenly as it tries to compensate for the change, which is not a desirable behavior.
3. The NxtMMX motor controller (and probably others) don't support this. We want the behavior of the tacho-motor class to be the same, at least as much as possible, regardless of the motor controller.

> If it is important, what is the best / fastest way to reset a motor position that just stopped? Is it enough to check motor state before and then change position?

As long as the motor is not running (`stop-mode` of `hold` counts as running), you should be able to change the `position`. As you have guessed, you can check the `state` attribute. If it does not have the `running` flag, then it is safe to change the `position`. If ramping is disabled (`ramp_*_sp` are `0`), then this should happen as soon as you send a `stop` command. From what you are saying though, it sounds like there is a delay, which sounds like a possible bug.

cavenel, 2015-05-13:
Thank you for this complete answer. After some tests, it looks like there is no bug and the position is changed as long as it is not in hold mode. I will close this issue, and reopen if I experience the problem in the future!

closed 2015-05-13 by cavenel