ev3dev/ev3dev #321: Friendly name / codename for ev3dev versions?

WasabiFan, 2015-05-09:
Now that support is being added for the Raspberry Pi and we are using the CKT kernels, the standard ev3dev kernel version names (what's currently reported by `uname`) are becoming fairly long and not very obvious. As evidenced by a few issues that I have seen over the last few days, some people are confused by the naming scheme and haven't been keeping track.

I get the need for these full technical names, but they aren't very user-friendly; (I'm not sure how one can refer to a specific version without using the whole version string). For some of our original releases, full version numbers were used for the kernel image names that were specific to ev3dev, but they were phased out in favor of release dates because it wasn't clear when our earlier images were released, which was needed to help support people with issues.

I think that we should consider finding a versioning scheme that we can use universally (how about a single number? :wink:) that we increment at every binary release. Maybe we could go so far as to use something more like a codename as is used for many other Linux distro versions.

So, @dlech and @rhempel, what do you think? I'm sure you guys have talked this over before, but I'd be interested in what your thoughts are on this.

dlech, 2015-05-09:
The single number is what I was going for on the last release. You still have to dig it out of the kernel version, but all the kernels will end with `-#-ev3dev-<flavor>`, so it's easy to dig out. Maybe we call the release `#-ev3dev` so it's a bigger part of the kernel name.

I think we should save the codename for a stable release. Although having codename == number could be bad because (for technical reasons) the number may have to increase for bug fixes in the stable release. Which just made me realize, there is a problem with my current system. If we have a stable branch and a development branch, we have a conflict if we have to increase the version number of both and we are using a single number.

Hmm, back to the drawing board...

dlech, 2017-08-12:
Good idea, but we've gone 2 years without doing anything or additional comments, so closing.

closed 2017-08-12 by dlech