ev3dev/ev3dev #322: Own terminal for applications / No LCD writing

theZiz, 2015-05-10:
Hi,

I am doing some LCD experiments right now and I ran into the problem, that I get the stdout output to the LCD, if I start it with the Brickman program starter. Even without writing to stdout I still get some ^[[D and similar output, when I press the buttons.

I read here (https://github.com/ev3dev/ev3dev/wiki/Using-the-LCD), that I need to get my own console, but the example link doesn't work anymore.

So my question is: In which file do I have to look, to get an idea?

Thanks!
Ziz

theZiz, 2015-05-10:
Okay, I found this one: https://github.com/ev3dev/brickman/blob/ev3dev-jessie/src/main.vala#L43 Seems to be, what I did search.

closed 2015-05-10 by theZiz

dlech, 2015-05-10:
When running a program from brickman, it will open a new virtual terminal (vt) for you, so including this in your own program is redundant. To run your program from an SSH session, you can use the `openvt` command . Since you are running it remotely, you have to run it as root. So what usually do is

    sudo openvt -s -w -- sudo --user <username> -- <program>

See #260 for more info.

Running this way has the advantage that if your program crashes, it will switch back to the virtual terminal that brickman is running on since the switcher is a separate program.

Also, another option is to use the `conspy` command. It works like this...

* Run `sudo chvt 2` in a SSH terminal. This will switch the display on the EV3 from brickman to a login prompt.
* Run `conspy` in the SSH terminal. You will now be able to control the display on the EV3 remotely.
* Log in and do whatever you want. If you switch the display to graphics mode, `conspy` will not be able to show you the screen, it is text only.
* When you are done, press ESC 3 times to close `conspy`.

theZiz, 2015-05-11:
Yeah, you are right, I found the wrong code. :\

All in all this is very hacky and not very straight forward. Wouldn't it be easier to get an option in brickman to not start a virtual terminal for an application? Like in the starter settings of Ubuntu and Co, where you can choose "run in a terminal"?

dlech, 2015-05-11:
> All in all this is very hacky and not very straight forward.

It is. There are not really any common real-life cases where you want to start a program from a remote terminal that runs on the local console. Also, it has been about 20 years since people were writing graphical applications for console, so there has not been much innovation in that area.

> Wouldn't it be easier to get an option in brickman to not start a virtual terminal for an application?

That is the part of #129 that we have not implemented yet.

theZiz, 2015-05-11:
I am not totally sure, whether you got, what I want. I am not very used to virtual terminals and such stuff, but with remote terminal you mean e.g. an ssh shell? However my problem is not related to such remote terminal, but to start applications with brickman without any ssh session.

However for now I just switch to graphic mode for the terminal and try to not write programs, which segfault. ;)

dlech, 2015-05-11:
> but with remote terminal you mean e.g. an ssh shell?

Yes.

> However my problem is not related to such remote terminal, but to start applications with brickman without any ssh session.

This is why I mentioned #129. The intention there is to make something like .desktop files where you can specify more about how an application should be run from brickman.

I just mentioned the other stuff because it is useful for testing and debugging your programs.

theZiz, 2015-05-11:
Ah, thank you. I didn't run into any problems with ssh - until I tried to use the buttons with brickman still running in the background. So thanks for this, I will try it out soon. :)