ev3dev/ev3dev #323: Help with new tacho-motor changes

sharived, 2015-05-11:
I need to update my code for tacho-motors and I'm struggling trying to figure out all of the changes.  I see that some attributes are gone - run_mode, pulses_per_second.   Is there a document describing the changes?  Or some examples you can provide?  In particular I'm wanting to get https://github.com/topikachu/python-ev3  working again.

dlech, 2015-05-11:
Check out <http://www.ev3dev.org/docs/tutorials/tacho-motors/>. It walks you through the "new" tacho-motor class.

dlech, 2015-05-11:
Also there is some discussion in #282.

sharived, 2015-05-11:
Great.  Thanks!  I was here http://www.ev3dev.org/docs/drivers/tacho-motor-class/ and didn't know about the tutorial.  Maybe you can add a link.

closed 2015-05-11 by sharived

dlech, 2015-05-11:
> Maybe you can add a link.

Or you can. :wink: There should be an "Edit on GithHub" link at the top of the page.

sharived, 2015-05-11:
Will do.