ev3dev/ev3dev #325: brickman networking screen stays "loading"

topikachu, 2015-05-12:
Just same as #196

But this happens on ev3dev-jessie-2015-05-01

dlech, 2015-05-12:
Exactly the same? So output of `systemctl status -l brickman` is the same? And restarting brickman with`systemctl restart brickman` brings the network menu back?

topikachu, 2015-05-13:
well. I have no network. how to input these commands? By a keyboard?

dlech, 2015-05-13:
It sounds like this is on a fresh install. If rebooting does not fix it, then you might try writing the image again. If that does not fix it, then you can attach a keyboard to the EV3 and press ALT+CTL+F6 to get a terminal and try to see what is going on.

topikachu, 2015-05-17:
well, it seems my sd card is corrupted. I use a new one and it's ok now.

closed 2015-05-17 by topikachu