ev3dev/ev3dev #328: Brickman wifi setup timeout

janorn, 2015-05-17:
Hi,

While trying to connect to my wifi with a rather long password Brickman will timeout half way through the password typing. The keyboard is rather slow to use. Please increase the wifi timeout.

I.e I do the following in Brickman.
Power on Wifi
Start scanning
Select my network
Connect
Start typing.

<timeout> and it will never finish even if I enter the correct password and type accept.

I use the latest release https://github.com/ev3dev/ev3dev/tree/ev3dev-jessie-2015-05-01

dlech, 2015-05-17:
How long would you say that it takes you to type in the passphrase?

I think the current timeout is set at 3 minutes, but I haven't actually tested it.

Alternately, you can enter your passphrase using `connmanctl`. You will have to setup a USB or Bluetooth connection first and open an terminal via ssh. Then see the section about "Connecting to a protected access point" [here](https://wiki.archlinux.org/index.php/Connman).

janorn, 2015-05-17:
It seemed more like 120 sec. Perhaps twice as much? Or maybe if possible being reset each time a key is pressed?

laurensvalk, 2015-05-17:
I've experienced the same as @janorn. My password isn't too complicated, but I have to switch between capitals, numbers and signs a few times. Once I practice once or twice I can finish it in time.

I also found that if it times out the first time, trying again (now in time) and clicking connect (or accept?) fails. This might be a bug. After reboot and on the third time practicing with the keyboard it works fine :)

Although I now happily use Bluetooth and USB, I did find setting up WiFi through the brick menu the easiest option when I first tried ev3dev, since it doesn't require any PC side configuration.

dsharlet, 2015-05-17:
Has anyone tried to use a USB hub to use a wifi dongle and a keyboard at
the same time? I've thought about trying this, but I haven't had to type in
a wifi password enough times to actually bother trying it. Then, hopefully,
the wifi password would be remembered after moving the wifi dongle from the
USB hub to the brick directly.

janorn, 2015-05-17:
I got it working with connmanctl. My pw is long and a lot of switching so I had no chance to finish in time.

dlech, 2015-05-17:
Sounds like we should increase the timeout to 5 minutes maybe?

And yes, using a keyboard with a USB hub (or Bluetooth) should work as well.

Anyone want to add a page on the website about these alternatives?

janorn, 2015-05-17:
5 minutes should do.

theZiz, 2015-05-17:
Fyi: I had the same problem. And always when I made an error while typing, I started from the beginning to avoid the timeout.

I didn't even know, that I could have used an usb keyboard instead...

Two thoughts to this topic:
a) Why is there even a timeout? What is the use of it? What happens if I would need 20 minutes for entering my wifi password?
b) Imo this problem would get much smaller when the button input is not linked to the lcd refresh frequency anymore. ;)

dlech, 2015-05-17:
a) 

> Why is there even a timeout?

Because it is part of ConnMan (which manages network connections).

> What is the use of it?

Because sometimes things just don't work out like they are supposed to. If you didn't have a timeout, then the programs would get stuck waiting forever for something that is not going to happen.

> What happens if I would need 20 minutes for entering my wifi password?

Then you should probably use one of the alternative methods.

b) Pull requests are welcome. :wink: (Or at the very least, open a new issue for it.)

theZiz, 2015-05-17:
> > b) Pull requests are welcome. :wink: (Or at the very least, open a new issue for it.)

I will definitely have a look at this, I didn't compile the kernel just for fun. ;)
(Sorry for offtopic)

dlech, 2015-05-21:
Timeout has been extended to 5 minutes in brickman v0.6.1

ensonic, 2015-09-22:
I think there are still issues. I am not even half through with entering the wifi password when I get this error:
https://drive.google.com/file/d/1TwPluL6EDOQoMjiu9SxAZPOivw7u2cafvw/view?usp=sharing
You get the same error, if you cancel.

I've tried several times to enter the pasword, but I get the timeout error after ~ 2 min.
https://drive.google.com/file/d/1GRb4Rx7EtbG5UOWNzHLptNj-ViQj7WWXcQ/view?usp=sharing
Why is there a timeout after all. Can't we just give the user all the time in the world to enter the password? To make it worse, after the timeot, we let the user continue to edit the password, press okay and select 'accept'. But then nothing happens.

What puzzles me even more is that if I look at the card on my linux machine, there is 
/var/lib/connman/wifi_e894f616f7b6_574c2d636c6f7564646576696365_managed_psk/settings
with the correct passphrase entered. Still it won't connect the wifi.

dlech, 2015-09-22:
Interesting. The first one is a new error, but we did just get a new version of ConnMan. I think there is another place where I need to change the timeout as well.

rhempel, 2015-09-22:
Stephan has a good point - we can probably make the timeout longer (or infinite) and presenting the last known password helps if there's just a typo.

dlech, 2015-09-22:
Yes, we've been discussing this on irc :wink:

dlech, 2015-09-22:
I found where the 2 minute timeout is coming from. It is in ConnMan. You can adjust it in `/etc/connman/main.conf`.

dlech, 2015-09-22:
I have fixed the default setting for ConnMan in `ev3dev-connman-config 1.1.0`.

I'm calling this issue fixed. We can start a new issue to discuss ways to actually improve the process.

ensonic, 2015-10-05:
Btw. this still happens, but I am fine to keep this closed:
1.) I have a super complicated wifi key phrase (alternating oder of number, specation chars and upper/lowercase chars) - that is I can't enter two chars without having to switch the keyboard-tab.
2.) Now I can keep typing and press connect afterwards and I don't have to enter the phrase again.

dlech, 2015-10-05:
Which error message are you getting? At least 2 is an improvement.

ensonic, 2015-10-05:
Still the "timeout" and I don't believe that I already spend 5 minutes when it fired.

dlech, 2016-05-31:
Don't know how I missed this before, but I have now properly extended the timeout to 5 minutes.

dlech, 2016-06-17:
Fix has been released in `brickman 0.8.0`

closed 2016-06-17 by dlech

khawks, 2020-01-31:
I am browsing these issues because I cannot seem to get brickman to connect to a hidden wifi network using a wpa_supplicant.conf file in /etc/wpa_supplicant/   
I have scan_ssid=1 along with the correct SSID string and PSK string.
Does connman override this type of setup.
My goal is to image 12 8GB micro cards that will connect to our hidden network, without students having to configure any network settings on the EV3.

dlech, 2020-01-31:
ev3dev uses ConnMan for network management. Don't touch `/etc/wpa_supplicant` or things will break.

Use `conmanctl` to setup a network.

Once you  have one EV3 setup, you should be able to copy the `/var/lib/connman/` directory to the other EV3s to copy the configuration.

khawks, 2020-01-31:
Thank you for the response David:
I would like to edit a file that stores my wifi login stuff using nano on another Debian box.  Without having to deal with entering data via the ev3 brick itself. “Profile settings are stored in directories under /var/lib/connman/ by their service name.”
This link is the closest I have come across but does not explain what should be a simple thing to do? (as in wpa_suppliant.conf for other devices)
https://www.toradex.com/community/questions/26797/prebuild-wifi-configuration-for-auto-connect-to-ne.html

thank you for your time!
Kirk.

Kirk Hawks
Math & Science @PCA
khawks@psdschools.org<mailto:khawks@psdschools.org>
970 490-3122

dlech, 2020-01-31:
It looks like we have connmand 1.36 on ev3dev-stretch, so it should be possible to create a provisioning file:

https://manpages.debian.org/testing/connman/connman-service.config.5.en.html