ev3dev/ev3dev #329: EV3 bytecode interperter/LEGO VM compatibility layer

dlech, 2015-05-19:
I discussed this with @botbench a while back and @laurensvalk just recently, so I figured I better start an issue so that all the ideas are in one place.

### Overview

The idea is that we should write a virtual machine/interpreter for the machine language used in the official LEGO software for EV3. If we do this, then applications that work with the official LEGO firmware and its close relatives like the firmware used by RobotC and LabVIEW could also work with ev3dev.

There are a number of 3rd party programming languages/environments that could benefit from the advanced features of ev3dev, like better Wi-Fi support, webcam support, RPi/BrickPi support, etc. The ones I know of are EV3-G (official software from LEGO), EV3Basic, RobotC, LabView.

There are also a number of applications that remotely control the EV3 using the same bytecodes. EV3 Commander (official software from LEGO), Simulink, MonoBrick remote control library, leJOS remote control library. All of these could work with ev3dev too.

### Implementation

For the actually implementation, I think it could easily be done using [ev3devKit](http://www.ev3dev.org/ev3devKit/ev3devkit/index.htm) (the same library used by brickman). It can be a daemon the just runs in the background and can be started and stopped like any other service. It should be fairly straight forward to reverse engineer the lms2012 code to figure out what bytecodes are needed and how they are supposed to work.

We could add an option to brickman that is a checkbox that says something like "Enable LEGO VM compatibility layer" to start and stop it.

USB: We currently have the options of off, CDC or RNDIS for network communications via USB. We can easily add a new option that enables an HID connection that will create a compatible connection. If we want to do both at the same time (i.e. have a network connection using CDC or RNDIS and a LEGO compatible HID connection), it will take some more work. It shouldn't be too bad though because the Linux gadget driver for USB already supports configfs, which makes implementing the combinations of these much easier.

Bluetooth: We currently have the issue/feature that the Bluetooth MAC address is different when using ev3dev when compared to the official LEGO firmware. This comes from the fact that the version of `hciconfig` that comes from Debian does not implement the feature to specify a MAC address for the built-in Bluetooth chip in the EV3. As a result, when using ev3dev, the vendor portion of the MAC address is `00:17:E7` (Texas Instruments) instead of `00:16:53` (LEGO). This will be a problem for any program that looks at the MAC address to decide if this device is an EV3 or not.

File storage: We will need a special place to store files sent using the compatibility layer. I think makes the most sense to do this in `/var/lib/`.

### Further discussion

I've though a bit about how to make this actually work, but not much about how people would actually use it. So, I will leave this part of the discussion to others. Does this sound like something you would like to use? How would you use it?

theZiz, 2015-05-19:
I myself would not use it, but I am quite sure, that many switchers would like a possibility to use their lego vm programs for start. As you mentioned, even with using this you already profit.

To the bluetooth problem: Would it possible to fork the debian hciconfig and to add the needed changes and to make an ev3 debian package out of it?

dlech, 2015-05-19:
> Would it possible to fork the debian hciconfig and to add the needed changes and to make an ev3 debian package out of it?

Yes. We just need to find the patch that is used in the version from the LEGO firmware. Hopefully it is publicly available somewhere.

laurensvalk, 2015-05-19:
### Motive

I'd really like to see ev3dev providing support for the applications that work with the standard LEGO firmware. In particular, I'd be interested to have support for the official LEGO MINDSTORMS EV3 programming software made by National Instruments (NI). I think this could enhance the potential of ev3dev and even LEGO MINDSTORMS EV3 in general.

My main motive would be to have an environment for EV3 that's both powerful and easy to use for beginners. A large portion of the EV3 target audience are kids who are new to programming and robotics in general. Many of them will start with the EV3 programming language and stay there. Text programming languages are scary (compared to colorful drag and drop programs). Those who proceed to use a text programming language, will probably first transition to software with an easy to use IDE such as RobotC.

This is the path I've personally followed as well. I purchased my first Robotics Invention System (the RCX) when I was 13. I used the LEGO graphical language for quite some time and eventually started using NQC with the Brixcc IDE. I was a Windows user then, which is probably typical (or Mac, but not Linux) for the average MINDSTORMS target audience.

### The idea

While the LEGO EV3 language is powerful and a great way to learn programming, it doesn't quite enable everything the EV3 can really do. This is where I can see ev3dev come in. I would think of ev3dev supporting the NI software as a way to extend the possibilities of the EV3 for the 12+ user who has mastered the NI software but wants to do more. He or she has seen YouTube videos of robots solving rubik's cubes or sort LEGO elements using a USB webcam, but all those applications are beyond their reach because the NI software does not support them.

One might argue that such a user should consider using a fully functional Linux distribution like ev3dev, learn to use Linux, all the while learning a new programming language or two. I don't think this is very realistic, and it would certainly scare off most new users.

If ev3dev would provide just the standard functionality of the NI software, there would be not much benefit for the average user. Why bother updating the firmware if there's nothing to gain? Here's the cool part. Why not unleash the full capabilities of ev3dev from within the EV3 language?

This doesn't have to be as complicated as it sounds. My idea would be to provide the basic functionality of the LEGO VM, but with one crucial difference: The option to execute a shell script from the user program. Some ideas for the technical implementation of this are given below. 

### Usage scenarios

I would think of two types of scenarios that might emerge:

- The beginning user can try out basic and at the same time exciting Linux functionality, all from within a familiar environment. Consider something as simple as using espeak/aplay. How cool is it that they can make their robot actually talk? If this captures the interest of the user, he or she might want to learn more about Linux and unlock more possibilities.

- More experienced users, like those already using ev3dev, can create cool projects and just share the script, so that all of the users mentioned above can try out all your cool projects from within their own programs. The possibilities here are endless. Of course, you are not limited to using shell scripts---just start your C program from within the script. The options to get that file in programming language X or Y on the brick are also numerous (just wget it), and the average user wouldn't even need to see the technical details.

### General implementation considerations

- In my opinion, ev3dev support for the NI software doesn't imply that we have to make something that looks and feels exactly like the LEGO VM. If the users see a penguin and brickman at boot, I think that's totally fine. However, it would have to be easy to install (equivalent to a basic firmware update) and easy to program (plug the brick with usb, and press the start button in the NI software.)

- A first goal might be to provide the basic functionality that the EV3 NI software currently provides for the NXT. (For those who don't know, the EV3 software can be used to program the NXT brick.) This functionality includes support for the LEGO sensors that come in the box, as well as the standard blocks for sound, movement, math, etc. However, this does not include support for Bluetooth between bricks, daisy chaining, and bluetooth to program the robot. Only USB support is available.

- More advanced functionality can be added later, or left to the scripting option given below. The advantage of the latter method is that this project has a clear goal, and a more limited scope, making it easier to maintain. (On the other hand, if support for all features is provided, I suppose there's no longer any reason not to use ev3dev.)

### Implementation considerations for running shell scripts from the NI based executable

* The easiest option (also the most limited) would probably be to have the reader create or download a script text file, and download it to the brick using the existing option in the NI software. Then, all that's necessary to add to the NI software is a way to start that script. Here are two ideas to do that:

[... cut ...]  the EV3 language in general.

### Why this long story?

As some of you may know, I have written a beginner's book to LEGO MINDSTORMS EV3 and using the official EV3 programming language. As such, I've had the privilege of getting feedback about LEGO MINDSTORMS from people of all ages (5-88) and many different backgrounds. Kids (and parents!) are really learning about science and technology with this platform. But most of all, they're just having fun, and I think ev3dev can make it even more fun.

sharived, 2015-05-19:
Hi,
  I would like to support the Lego Ev3 software because I would like to get my camera supported by FLL.  I was just about to try and figure out how to get this done.  

@laurensvalk - when at the San Mateo Maker Faire this past weekend I got a lot of positive feedback about having [Snap!](http://snap.berkeley.edu/)  available for Ev3 so I just made that a reward for my [kickstarter](https://www.kickstarter.com/projects/robosnap/robosnap-vision-for-your-robot/) and I am planning to make that available outside of kickstarter as well.  There are lots of kids familiar with Scratch and I think Snap! is closer to a real programming language than the Scratch and the Lego NI software.  Would love to get your feedback.

Thanks,
Shari

WasabiFan, 2015-05-19:
I agree with all the posts above (wow @laurensvalk that's a long post!). I think that that "older kid" group where they are looking for more advanced things to do with their EV3s is a great target audience for ev3dev.

I'd be interested to see how you (@dlech) figure out how to map the LEGO format to ev3dev commands. I'd imagine that there will be some issues with translating specific functions that LEGO's software has into equivalent functionality on our side, and the efficiency (pre-translating their format vs. interpreting it on-the-fly) could become problematic. But I'm sure that you have thought through most of this and can pull it off!

I'd like to add that if we can get a system working that translates the official EV3 APIs into ev3dev motor driver commands, it would be nice to build a desktop app around it that would allow people to take their existing EV3 programs and convert them to ev3dev-compatible scripts. A program like this could translate EV3-G programs into some sort of a common internal format, and then output code in a user-chosen language (any language that has a supported language wrapper written for it as well as a definition set for the APIs exposed by that library). Adding something like this would completely bridge the gap between LEGO's EV3 firmware and ev3ev: you can run your EV3 programs on ev3dev, you can create native ev3dev programs from existing EV3 code, and you can write programs from the start for ev3dev.

So, I think I support the idea of building it off of ev3devKit as a standalone program so that the code can be easily used in a client-side scenario.

As a final thought, I think that adding this VM support would probably be a good piece to tie into the full brickman program selection system. Having a single front-end place to run both EV3-G programs and custom ev3dev scripts would really be a big milestone for this project.

laurensvalk, 2015-05-20:
>  @sharived wrote: There are lots of kids familiar with Scratch and I think Snap! is closer to a real programming language than the Scratch and the Lego NI software. Would love to get your feedback.

I think many programming languages can be used to get started with programming. I actually like the NI software a lot, and I would argue that it is also a 'real' programming language, but that is just my personal preference. But my main reason for supporting the NI language is simply because it is what most people are using. Support for the default programming language makes the transition for beginning users a little easier.

From a technical point of view, I suppose it would be easier to implement Snap! or Scratch right on ev3dev, rather than on top of the LEGO style VM discussed here.

> @WasabiFan wrote:  I think that that "older kid" group where they are looking for more advanced things to do with their EV3s is a great target audience for ev3dev.

Who knows --- when their number exceeds the number of "advanced" ev3dev users, the brickman checkbox that activates the LEGO style VM might need to be checked by default :)

> @WasabiFan wrote: a program like this could translate EV3-G programs into some sort of a common internal format, and then output code in a user-chosen language (any language that has a supported language wrapper written for it as well as a definition set for the APIs exposed by that library).

This might be quite challenging, due to the way the NI language is designed. Data is carried between programming blocks with data wires, such that data 'flows' from one block to the next. When converting this to code, these would probably become temporary (untitled) variables. This might work technically, but it wouldn't show the user how to write a true text based program. I think languages like Scratch would be more easily converted to a text language, since they use a similar paradigm for data flow.

dlech, 2015-05-20:
> Consider something as simple as using espeak/aplay.

I know many kids who would spend all day playing with Mindstorms if it had a "speak" block. Oh, the silly things it would say.

> ev3dev support for the NI software doesn't imply that we have to make something that looks and feels exactly like the LEGO VM

Yes, we only need the byte code interpreter part, we don't need to run the UI stuff from lms2012.

> running shell scripts from the NI based executable

Brilliant!

> I'd be interested to see how you (@dlech) figure out how to map the LEGO format to ev3dev commands.

We have modelled the sensor and motor classes based on LEGO's design, so I don't anticipate having any serious issues with this.

> A program like this could translate EV3-G programs into some sort of a common internal format, and then output code in a user-chosen language

Instead of trying to reverse engineer the XML files that make up an EV3-G program, I would suggest writing a disassembler for the VM byte codes. The lms2012 software already has an intermediate language that you can use with a java program to compile to machine code.

> I think languages like Scratch would be more easily converted to a text language

Snap! already has this feature. It can be converted to Python, JavaScript and one or two other languages, IIRC.

dlech, 2015-05-28:
I've been tinkering with the lms2012 code a bit this week to see just what exactly it will take to accomplish creating something similar in ev3dev. Here is what I have come up with so far.

1. I realized that any TCP/IP connection is seen as "Wi-Fi" by the LEGO software, so we can get this off the ground without having to make look-alike USB and Bluetooth connections as described above. We can use the USB and Bluetooth network connections as we do now in ev3dev. It will just say that the connection type is "Wi-Fi" in the LEGO software.
2. I realized that this is going to take massive amounts of coding. There are around 500 bytecodes that need to be implemented (not including the system commands, which adds about 10 more). Even if we leave out the Bluetooth and daisy chaining stuff as suggested, that is only a small percentage.
3. I created some [tools for reverse engineering EV3 bytecodes](https://github.com/ev3dev/lms-hacker-tools). These should be helpful for many things including this issue.
4. I think we can hijack the daisy chain feature to do some cool stuff. For example, you could say that the sensors connected to daisy chain layer number 2 are the sensors on a HiTechnic sensor multiplexer. Then you could actually see these sensors in the sensor browser in the LEGO software. Same goes for motor multiplexers.
5. I have got far enough that I have the LEGO software thinking that a python program running locally is an EV3 connected via Wi-Fi (after crashing the software many times). Due to the fact that the bytecodes require lots of pointers and C type stuff, I'm going to have to abandon python. Even using ctypes, it's too much trouble.

botbench, 2015-05-29:
I think it would be "simpler" to create a compatibility layer on the OS and
make some minor changes to the VM, than to recreate a whole new VM.

The interface down to the kernel for the LEGO VM is not that complicated,
certainly much easier than the bytecode interface towards the programming
environment.

= Xander

| I'd Rather Be Building Robots
| Current projects and ramblings: botbench.com
| Google+: gplus.to/XanderSoldaat

dlech, 2015-05-29:
> I think it would be "simpler" to create a compatibility layer on the OS and make some minor changes to the VM, than to recreate a whole new VM.

When I read this, I though to myself, "Wow, David, you really like doing things the hard way, don't you?" In fact, I have recently read a good article about how programmers like to throw everything out and start from scratch and how foolish that is. But, the more I think about it, for this particular case, I'm not so sure. Of course, reusing the existing code would eliminate the need to re-implement all of the trivial byte codes, but on the other hand, we wouldn't be able to use ev3devKit, which means recreating lots of other work that I have already done.

Also, I'm thinking about what it would take to create the compatible character device drivers that would interface with our existing drivers. It's not what I would call "simple". All of the ev3dev drivers would have to be modified to allow internal kernel access by other drivers. Also, to make this configurable where it would work on other systems like the BrickPi would be tricky. To me, this sounds like almost as much work as recreating the VM.

Although, who says we have to write it as kernel drivers. We could replace all of the `d_*` drivers from lms2012 with a DBus service. Then we could easily use ev3devKit. Hmm... may be on to something here.

laurensvalk, 2015-06-04:
The following is a more visual description of the method I described above, considering how to start scripts from within the EV3 program.

I realized we may not need to do any customization on the NI software side of things at all, even if we want to run shell scripts that have both inputs and outputs. The result can still look quite elegant. To be clear, the following is all hypothetical - it is just one way to make this possible. 

[Background information: The cyan colored blocks in the programs below are so called 'My Blocks'. The user can use these to make custom functions based on selected inputs or outputs. Normally, My Blocks are composed of other regular programming blocks (Just like you make functions in any other programming language.) Above and below I explain how this might be extended to run ev3dev shell scripts.]

The simplest way is to run a script with no inputs or outputs. I would imagine running the script called "hi.sh" as follows:

[image]

That is, the motor moves, the script specified as "hi.sh" runs, and then some sound plays. The "RunScript" block would look something like the following. This is just a FileAccess block that opens the specified file for reading. The ev3dev LEGO interpreter could interpret the action of reading the .sh file as executing it instead. (And return status info instead of the file contents.)

[image]

This can quite easily be extended to scripts with both inputs and outputs. Suppose I have the following EV3 program. This program reads a sensor (yellow block), does something super special with ev3dev (maybe using a custom camera) using the AwesomeFunction block (cyan). This function takes the sensor value as an input, as well as some other parameter (with value 5). It does all this by running the specified script ("hi.sh"), which is stored on the brick. When it's finished, it displays the result of the function (yellow wire) on the screen. It also says "Detected" or "Error", depending on whether the AwesomeFunction block ran successfully (green wire). 

[image]

The contents of the AwesomeFunction block are as follows. You can see 3 My Blocks. The WriteInput block contains blocks to write the inputs to a default text file (input.txt). We might define some standard format for this. Then the RunScript block executes the script. The ReadOutput retrieves any output from the script by reading the output.txt file.

[image]

My Blocks like these can easily be shared between users, so running scripts in this way should be fairly trivial for beginning users. We can also make a nice collection of example blocks on the ev3dev.org page. A simple example is a Speak block that runs espeak and aplay with the specified input string.

ameysutavani, 2015-06-06:
I think there is another option available:

Instead of trying to keep compatibility between LMS software and ev3dev (bothways), why not design a programming IDE program that internally functions on ev3dev but looks like the NI software?

Ofcourse, this will be easier said than done. But I think this will ease off the compatibility constraints.

Those users who want to explore ev3dev's additional functionality, but want to keep the ease of NI like block based programming can use this alternate "ev3dev desktop IDE" (If I may) to get things done.

ensonic, 2015-06-30:
Can't we use the original vm from https://github.com/mindboards/ev3sources-xtended/tree/master/ev3sources/lms2012/lms2012/source?

dlech, 2015-07-01:
> Can't we use the original vm...

@botbench suggested this already. We can't use it without modifications, but we certainly can used it if someone is willing to put in the effort to adapt it to ev3dev.

jabrena, 2015-09-10:
Was it scheduled?

ensonic, 2015-09-10:
@jabrena sorry, what do you mean? Related to the feature or how the VM works? As @dlech said, we need someone to give this a shot and work on it. To my knowledge no one is one it right now.

jabrena, 2015-09-10:
I am interested to collaborate some hours per week, it is necessary

dlech, 2015-09-10:
That would be super! No one has started working on this, so if you don't know where to start of if you have any other questions about anything, you can ask here or on IRC.

ensonic, 2015-09-11:
@jabrena super cool! I think there first step would be to extract the vm code from the mindboards git and try to build in in brickstrap (arm cross compilation env). This will most likely fail for a while. We need to indentify the parts of the vm code that access the hardware and then port them one-by-one to use the ev3dev drivers and interfaces.

jabrena, 2015-09-11:
Nice! At home I have a Mac, I will begin this weekend.
We are in contact.

rhempel, 2015-09-11:
You don't actually need to build it in brickstrap - just the plain old ARM cross compiler will work too. I usually set up a VM - instructions here [http://www.ev3dev.org/docs/devtools/setting-up-the-ev3dev-build-ecosystem/#building-the-kernel]. The previous sections on that page tell how to set up the VM - and I have a video on YouTube as well [https://www.youtube.com/watch?v=3YevPzDQ5ME]

jabrena, 2015-09-11:
I will follow this guid. Many thanks @rhempel 

I think if a user could execute a EV3G program on Debian could be great! imagine a script running in another language (node.js for example) which it writes a file in the file system and later a EV3G program could read that file. I think that easy way could increate the posibilites of this programming environment for example.

laurensvalk, 2015-09-11:
> I think if a user could execute a EV3G program on Debian could be great!

[Here] (https://github.com/ev3dev/ev3dev/issues/329#issuecomment-108919405) are some ideas for the implementation on the EV3 side of things.

I'm happy to help with development as well. I'm not sure where to start, but when it gets going, I'm sure I can help expanding and testing the project. 

I can set up some typical EV3-G use cases with source code we can use for testing. I could also provide building instructions for a simple vehicle so we all have the same test setup.

By the way, we may also need to consider whether we want to make the VM work exactly like the LEGO VM (including bugs) or not. That is, do we want existing programs to run the same on ev3dev as they do on the LEGO stock firmware?

First, I think it would be quite a challenge to make everything work exactly the same, since execution time will likely be slightly different (faster?).

Also, I'm not opposed to actually improving things. One example of a LEGO VM bug is in the Invert Motor Block, which is used to select the positive direction of a motor, so you can invert it if you mount the motor upside down (this functionality also exists in ev3dev, I think). In the LEGO VM this works, as long as you use a single motor. Once you use a block that controls 2 motors, the inverted setting is ignored, for some reason.

dlech, 2015-09-11:
@jabrena, just in case it wasn't clear, you don't need to follow the *entire* guide from @rhempel, just the section that he linked to. No need to set up the web server or package mirrors.

Also, I've created a new repository at <https://github.com/ev3dev/lms2012-compat>. You should fork this and work from there. I've merged <https://github.com/mindboards/ev3sources/tree/master/lms2012/lms2012> into it, which should be the bulk of the VM. Like @ensonic said, don't expect it to compile.

laurensvalk, 2015-10-21:
>  I could also provide building instructions for a simple vehicle so we all have the same test setup.

Now available [here] (http://robotsquare.com/2015/10/06/explor3r-building-instructions/). Easy to replace EV3 brick, with accessible SD and USB slots. Minimum building time.

[image]

Can also be used for other generic ev3dev stuff, BrickPi/PiStorms testing and so on.

dwalton76, 2016-01-10:
Regarding bluetooth and robot commander, I jumped through few hoops but still cannot get robot commander on my iphone to pair with ev3dev.  I did:

```
# Added /etc/machine-info
PRETTY_HOSTNAME=EV3

# Edited /etc/bluetooth/main.conf
Name=EV3

# Use bdaddr to change bluetooth MAC:
http://blog.petrilopia.net/linux/change-your-bluetooth-device-mac-address/
sudo apt-get install libbluetooth-dev
wget http://www.petrilopia.net/wordpress/wp-content/uploads/bdaddrtar.bz2
tar -xvf bdaddrtar.bz2
cd bdaddr
make

# To change the address
robot@ev3dev[bdaddr]# ./bdaddr 
Manufacturer:   Texas Instruments Inc. (13)
Device address: D0:07:90:B6:30:8C
robot@ev3dev[bdaddr]# 
robot@ev3dev[bdaddr]# sudo ./bdaddr -i hci0  8C:30:B6:53:16:00
[sudo] password for robot: 
Manufacturer:   Texas Instruments Inc. (13)
Device address: D0:07:90:B6:30:8C
New BD address: 8C:30:B6:53:16:00

Address changed - Reset device now
robot@ev3dev[bdaddr]# 
robot@ev3dev[bdaddr]# 
robot@ev3dev[bdaddr]# sudo service bluetooth restart
robot@ev3dev[bdaddr]# 
robot@ev3dev[bdaddr]# ./bdaddr 
Manufacturer:   Texas Instruments Inc. (13)
Device address: 00:16:53:B6:30:8C
robot@ev3dev[bdaddr]# 
```

bdaddr is a little buggy, the first time I did this I ran "sudo ./bdaddr -i hci0 00:16:53:B6:30:8C" and the "New BD address" looked correct but after restarting the bluetooth service the MAC was 8C:30:B6:53:16:00. 

Robot Commander still doesn't see my ev3 though.  Who knows they may be looking for some specific signature.  Any ideas...other than hacking together a bluetooth sniffer to see how robot commander interacts with the normal EV3 software?

dlech, 2016-01-10:
Further iPhone bluetooth discussion should be directed to #500.

ValorNaram, 2016-05-17:
The NI IDE for kids by lego creates files in a different format.
The only thing, we have to do, the files to study and to interpret it and to convert it in C or python code.
If we make it in Python, I can help!

dlech, 2016-07-03:
Good news for those interested in this issue. I have been privately contracted to get lms2012 working on top of ev3dev on a new hardware platform for BeagleBone based on the FatcatLab EVB.

If you are interested in following the progress, please watch the [lms2012-compat repository](https://github.com/ev3dev/lms2012-compat).

Feel free to open any issues there for ideas you have or bugs in the official LEGO software that you would like to see fixed.

The contract does not include getting this working on the EV3, so if anyone wants to help make that happen, I will be glad to review your pull requests.

WasabiFan, 2016-07-04:
Ooooh, so secretive 😁 A "private contract" related to a "new hardware platform" -- I'm interested to see what that turns into.

dlech, 2016-10-11:
This is substantially complete. If anyone has a FatcatLab EVB and a BeagleBone and wants to test, let me know.

closed 2016-10-11 by dlech

dlech, 2016-12-31:
There is now a version of lms2012-compat that will run on an EV3. Any testing would be appreciated. Here is what you need to do to try it out...

* Grab the SD card image [here](https://github.com/ev3dev/lms2012-compat/releases). 
* Flash it, boot it up and get a remote terminal following the usual [getting started](http://www.ev3dev.org/docs/getting-started).
* If you want to use bluetooth, turn it on in Brickman first
* In the remote terminal, run `sudo lms2012`.
* Use it with the Official EV3-G software.
* Report any problems, suggestions, etc. to the [lms2012-compat issue tracker](https://github.com/ev3dev/lms2012-compat/issues) rather than here in the main ev3dev issues.