ev3dev/ev3dev #330: Increasing Font-size on EV3's Display

LegoMindstormsRobots, 2015-05-20:
Hi on-and-all,

I'm old and going blind, deaf, insane......

Can someone please help me by point out how I may Increasing Font-size on EV3's Display. I find it very difficult to read in its current format (If LEGO has spent a cent or two for a LCD Backlight, things might have been improved).

cheers

Ray McNamara
www.LegoMindstormsRobots.com

dlech, 2015-05-20:
The fonts are hard coded, so there is not a way to change them easily.

You can fork the brickman repository, change the font sizes, fix any issues that arise because of this and submit a pull request. Or wait for someone else to do this. Or get a magnifying lens. Or install a backlit LCD.

LegoMindstormsRobots, 2015-05-20:
Thanx dlech for your prompt reply.

But I am using a large 5x Diopter Magnifying Lamp. It is the same on I use for my SMD electronics work.

[image]

I guess I'll have to wait, as at this stage my programming isn't up to scratch.....

cheers

Ray

dlech, 2015-05-20:
You are not the first person to ask for a larger font, just the first to take the time to open an issue for it. Let's leave this open and I am sure someone will eventually get around to changing it.

P.S. It looks like your image didn't finish uploading.

sharived, 2015-05-20:
The part I always have reading is the IP address at the top of the screen.  Maybe that can be moved down a few pixels.

dlech, 2015-05-20:
Moving the IP address is a different issue from changing the font size, so please open a new issue for that.

sharived, 2015-05-20:
Sorry - moved to issue #332

dlech, 2015-12-16:
Fix has been released in brickman 0.7.0. More casual users will want to wait for the next image release before updating. See the [announcement](http://www.ev3dev.org/news/2015/12/15/Package-Release/) for more info.

closed 2015-12-16 by dlech

WasabiFan, 2015-12-21:
Personally, I think that the new font size is too unwieldy on the EV3 display -- I admit that the previous one could get a bit hard to read, but this new one is big enough that buttons are pushed below the fold and you have to scroll to see all the options in most views. This might just be something I have to get used to, though.

Taking a brief look through the brickman code, I see that there's a method that returns the default -- could it be modified so that the font size is a setting? More simply, I think it would be great if the font could be a size in between the current and the previous; but I expect that it's this size because letters don't align well with other sizes.

dlech, 2015-12-21:
Please open a new issue.

WasabiFan, 2015-12-21:
OK... It seems like a discussion on the amount we increased the font size is a valid topic for "Increasing Font-size on EV3's Display" :wink:, but I'm happy to open a new issue if you think that's best.

dlech, 2015-12-21:
It is always best to open a new issue instead of commenting on a closed issue unless you are 1) confirming the fix works or 2) proposing that the issue should be reopened because it is not actually fixed.