ev3dev/ev3dev #331: Support for Fatcatlabs's EVB BeagleBone Cape

dlech, 2015-05-20:
As suggested in #312, it should be easy to adapt ev3dev to [Fatcatlabs](http://www.fatcatlab.com/)'s EVB BeagleBone Cape (recently on [KickStarter](https://www.kickstarter.com/projects/fatcatlab/evb-replace-the-brain-of-your-lego-ev3-with-beagle)).

They are not currently available for purchase, but if anyone bought an extra on KickStarter they want to donate, I have no doubt we can make it work. Otherwise, we'll just have to wait for if/when they are available for general purchase.

dlech, 2015-06-14:
Looks like these are on sale now. As an alternative to one or two people donating the hardware, we have setup an account on bountysource. You can make a pledge towards this specific issue at <https://www.bountysource.com/issues/16245764-support-for-fatcatlabs-s-evb-beaglebone-cape>. If we get $175 worth of pledges, we will purchase the hardware.

dwalton76, 2015-09-28:
Bumping this thread...I have an EVB, I would love to run ev3dev on it :)

dlech, 2015-10-30:
Got an EVB in the mail yesterday! Now I need to order a BBB to go with it.

dlech, 2016-01-26:
http://0.0.0.0:4000/news/2016/01/25/evb-ev3dev-jessie-2016-01-25-release/

closed 2016-01-26 by dlech

rhempel, 2016-01-26:
Dead link :-)

rhempel, 2016-01-26:
David - take a well deserved break - you've done a hell of a job!

dlech, 2016-01-26:
Ha! that was my local preview. 

http://www.ev3dev.org/news/2016/01/25/evb-ev3dev-jessie-2016-01-25-release/

rhempel, 2016-01-26:
Looks like a proper mad scientists bench - the holes in the plastic are nothing a drill can't fix ;-)