ev3dev/ev3dev #332: Hard to read the IP address at the top of the brickman display

sharived, 2015-05-20:
Hoping we can move it down a few pixels.

dlech, 2015-05-20:
I am thinking that you have the same shadow at the top of the screen that I see on my EV3 that is making it hard to read.

I was going to suggest maybe moving the statusbar to the bottom of the screen, but then whatever else happens to be at the top of the screen will be hard to read, so this is probably not a good idea.

sharived, 2015-05-20:
RIght, I'm busy turning it different angles to see it.   And with the small lettering it is even harder.

dlech, 2015-12-16:
Fix has been released in brickman 0.7.0. More casual users will want to wait for the next image release before updating. See the [announcement](http://www.ev3dev.org/news/2015/12/15/Package-Release/) for more info.

Just making the font bigger seems to be good enough to make it more readable.

closed 2015-12-16 by dlech