ev3dev/ev3dev #333: Brickman back to terminal option

theZiz, 2015-05-21:
Hey,

today I ran into the problem, that while updating the brick my bluetooth connection was killed. Some minutes later brickman also exited, which was in fact quite handy, because I could connect a keyboard and continue working. However later brickman was restarted. At that point I would have liked an option to exit it or at least to open a new terminal inside it.

Would that be possible?

Greetings, Ziz

dlech, 2015-05-21:
Duplicate of #173.

Just like most other Linux systems, you can press `ALT`+`CTL`+`Fn` to open a new virtual terminal. Since we are using systemd, `tty6` is always guaranteed to give you a login prompt, whereas other ttys may be used for other things. So, `ALT`+`CTL`+F6` is the best bet. With the latest updates, brickman will always run on `tty1`, so pressing `ALT`+`CTL`+F1` will take you back to brickman.

We should probably add a page to the tutorials with this information.

theZiz, 2015-05-21:
Sorry, I didn't know what to search. Furthermore I didn't think of such an easy solution. Thanks!

closed 2015-05-21 by theZiz