ev3dev/ev3dev #334: Led names of Image release ev3-ev3dev-jessie-2015-05-20

theZiz, 2015-05-21:
I just did a upgrade via
apt-get update
apt-get upgrade
apt-get dist-upgrade
to the most recent version (from 20th may)
but the leds are still named
/sys/class/leds/ev3:green:left
instead of
/sys/class/leds/ev3-left0:green:ev3dev
as you stated. Or did you mean something else in the release notes?

Greetings, Ziz

BertLindeman, 2015-05-21:
Did you do a reboot after the installation of the new kernel?

theZiz, 2015-05-21:
Of course

BertLindeman, 2015-05-21:
:smiley:  I just updated my ev3 today and after reboot it shows here:
```
 uname -a
Linux ev3dev 3.16.7-ckt11-5-ev3dev-ev3 #1 PREEMPT Wed May 20 17:02:00 CDT 2015 armv5tejl GNU/Linux
12:48:03 root@ev3dev:~# ll /sys/class/leds/
total 0
lrwxrwxrwx 1 root root 0 May 21 11:47 ev3-left0:red:ev3dev -> ../../devices/platform/leds_pwm/leds/ev3-left0:red:ev3dev/
lrwxrwxrwx 1 root root 0 May 21 11:47 ev3-left1:green:ev3dev -> ../../devices/platform/leds_pwm/leds/ev3-left1:green:ev3dev/
lrwxrwxrwx 1 root root 0 May 21 11:47 ev3-right0:red:ev3dev -> ../../devices/platform/leds_pwm/leds/ev3-right0:red:ev3dev/
lrwxrwxrwx 1 root root 0 May 21 11:47 ev3-right1:green:ev3dev -> ../../devices/platform/leds_pwm/leds/ev3-right1:green:ev3dev/
```
This does probably not help very much. Sorry.

theZiz, 2015-05-21:
Brickman doesn't turn green when starting, so probably it tries to read the non existing leds, too. Maybe my upgrade failed. At one point I had to decide between a local file and a package maintainer file. Maybe I shouldn't have select the maintainer file. Furthermore I shutdown the brick while updating. Afterwards apt-get told me to use "dpkg somewhat -a" to continue and repair the installings. Maybe this failed.

I will do a full reflash. Did you just update via update/upgrade/dist-upgrade?

theZiz, 2015-05-21:
After a full refresh the leds are named correctly. Thanks. ;)

closed 2015-05-21 by theZiz