ev3dev/ev3dev #336: UART sensors not working with Image release ev3-ev3dev-jessie-2015-05-20

theZiz, 2015-05-21:
Hi,

In the chat someone stated, that the UART sensors (like the colour sensor) are not working with the most recent version. I don't know his name, he was called Guest76137, maybe he will react here and we get to know his name.

Anyway. I did update to the most recent version too with apt-get (+reboot) and the uart sensors stopped working here, too. I connected it to port4 and the directory
/sys/class/lego-port/port3/in4:ev3-uart-host
does exist, but the directory
/sys/class/lego-sensors doesn't.

Analog sensors and motors seem to work fine.

Greetings, Ziz

karlkappe, 2015-05-21:
Hi,

thanks theZiz for opening the issue.
Here is the output of:
dmesg | grep ev3
[    2.906515] systemd[1]: Set hostname to <ev3dev>.
[   13.599114] lego-port port0: Registered 'in1' on 'legoev3-ports'.
[   13.625285] lego-port port1: Registered 'in2' on 'legoev3-ports'.
[   13.643837] lego-port port2: Registered 'in3' on 'legoev3-ports'.
[   13.669145] lego-port port3: Registered 'in4' on 'legoev3-ports'.
[   13.701830] lego-port port4: Registered 'outA' on 'legoev3-ports'.
[   13.741274] lego-port port5: Registered 'outB' on 'legoev3-ports'.
[   13.756817] lego-port port6: Registered 'outC' on 'legoev3-ports'.
[   13.787644] lego-port port7: Registered 'outD' on 'legoev3-ports'.
[   14.033904] lego-port port1: Added new device 'in2:lego-ev3-touch'
[   14.156058] tacho-motor motor0: Registered 'lego-ev3-l-motor' on 'outA'.
[   14.160841] lego-port port4: Added new device 'outA:lego-ev3-l-motor'
[   14.211200] tacho-motor motor1: Registered 'lego-ev3-m-motor' on 'outC'.
[   14.229891] lego-port port6: Added new device 'outC:lego-ev3-m-motor'
[   14.266177] tacho-motor motor2: Registered 'lego-ev3-l-motor' on 'outD'.
[   14.296992] lego-port port7: Added new device 'outD:lego-ev3-l-motor'
[   14.705195] lego-port port2: Added new device 'in3:ev3-uart-host'
[   15.065869] input: EV3 speaker beep as /devices/platform/snd-legoev3/input/input2
[   15.410311] lego-port port0: Added new device 'in1:ev3-uart-host'
[   15.633680] lego-port port3: Added new device 'in4:ev3-uart-host'
[   24.445972] lego-sensor sensor0: Registered 'lego-ev3-touch' on 'in2'.

So it seems that the brick detect that there is something, but somehow the
/sys/class/lego-sensors/
only shows the analog touch sensor.
Also on the brick in the Device Browser -> Sensors 
only the analog one appers.

Greeting Karl

dlech, 2015-05-21:
Like @theZiz said on IRC, "nobody's perfect".  I was going to make a breaking change with port names in the kernel, but later decided against it. I reverted the change in the kernel before this release, but forgot to revert the change in the `ev3-systemd` package.

This is fixed in `ev3-systemd` version 3, so upgrade it an reboot and EV3/UART sensors will work again.

closed 2015-05-21 by dlech