ev3dev/ev3dev #337: Process spi0 using a steady 50-75% of CPU

laurensvalk, 2015-05-21:
With the kernel I built with yesterday's release (3.16.7-ckt11-6-rc1-ev3dev-ev3), the spi0 process is using a steady 50-75% of the CPU. This doesn't seem right, and I don't think I've seen it in previous builds.

This is persistent after rebooting. What is this process used for? Is it used to talk to a certain peripheral within the brick?

```
PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND                                                                                              
   36 root      20   0       0      0      0 R 74.7  0.0   7:47.52 spi0                                                                                                 
    3 root      20   0       0      0      0 S  6.3  0.0   0:40.46 ksoftirqd/0                                                                                          
  612 segway    20   0    4964   2508   2144 R  3.9  4.2   0:00.81 top                                                                                                  
  459 root      20   0   54924   8596   6776 S  1.2 14.5   0:35.63 brickman  
```

dlech, 2015-05-21:
`spi0` is communication with the anlaog/digital converter chip. This is why we have it polling at 100Hz instead of 1000Hz. In addition in the latest kernel release, I change `CONFIG_HZ` from 100 to 1000 which seems to have affected it as well. (Note: these are 2 completely different things that were changed, it just happens to be the same numbers which could be confusing).

So, with the latest kernel with `CONFIG_HZ` at 1000, I am seeing cpu usage of `spi0` between 9 and 10%. With your additional tweak to the polling time, this is why you are seeing higher usage.

dlech, 2015-07-07:
From @pabr on #351:

> I can confirm that ev3-ev3dev-jessie-2015-05-20 feels more sluggish (as experienced in a shell over ssh over usbnet/cdc) than the version I was using 6 months ago. Disabling legoev3_analog.ko (and therefore spi0) solves the problem.

> I am not sure why CONFIG_HZ=1000 increases spi0 CPU usage. Maybe it causes so much overhead that the spi0 tasklet ends up consuming a larger portion of what little CPU time is left. I will recompile with HZ=100 and report here if that helps figure out anything. Regardless, I don't think HZ=1000 was ever intended for embedded systems running at 300 MHz.

dlech, 2015-07-07:
@pabr, this is a better place to discuss spi CPU usage since #351 is really about memory usage.

I will turn `CONFIG_HZ` back to 100 in the next kernel release.

@laurensvalk, in your experiments related to #324, did you notice a difference between HZ=100 and HZ=1000?

pabr, 2015-07-07:
I just noticed an anomaly possibly related to CONFIG_HZ=1000:

> # cat /proc/version      
> Linux version 3.16.7-ckt11-5-ev3dev-ev3 (david@freyr) (gcc version 4.5.2 (Sourcery G++ Lite 2011.03-42) ) #1 PREEMPT Wed May 20 17:02:00 CDT 2015
> # cat /proc/cpuinfo      
> processor       : 0                                                             
> model name      : ARM926EJ-S rev 5 (v5l)                                        
> BogoMIPS        : 1495.04

I believe the correct bogomips value is 149.50. Maybe the CPU detection code assumes HZ=100.

This might cause udelay() (and all the other busy-waiting timing functions in the kernel) to waste 10 times more CPU than they should.

dlech, 2015-07-07:
Interesting. We are using a busy wait with spi0, so this sounds like a plausible reason that it is using more CPU with HZ=1000.

pabr, 2015-07-07:
I rebooted ev3-ev3dev-jessie-2015-05-20 without lpj=747520 on the kernel command-line in order to let init/calibrate.c estimate bogomips. This brings spi0 CPU usage back to 3% and everything feels more responsive. 

```
Hit 'l' to stop autoboot:
U-Boot > mmc rescan 0
U-Boot > run loadimage
U-Boot > setenv bootargs mem=${memsize} console=${console} root=/dev/mmcblk0p2 rw rootwait splash
U-Boot > bootm ${loadaddr}
```

dlech, 2015-07-07:
Thanks for working on this. It would have taken me a while before I remembered we have set the `lpj` parameter. The reason it is there is just because it was copied from the boot script of the official LEGO firmware. 

According to [this](http://elinux.org/Preset_LPJ), it only saves about 1/4 second during boot if HZ=100 and only 25msec if HZ=1000. Sounds like we should just removed this option. It is set in the [flash-kernel package](https://github.com/ev3dev/flash-kernel).

rhempel, 2015-07-07:
Yes, thanks for seeing this anomaly! Now if we can confirm that HZ=1000 is a much better response form some sensors, or if there's an optimum value between 100 and 1000 then we'll be all good!

pabr, 2015-07-08:
Here is an attempt to measure the cost of increasing HZ from 100 to 1000 in terms of CPU throughput (not latency or scheduling accuracy). This could be wrong because I am comparing 3.16.7-ckt11-5 (HZ=1000, compiled with gcc-4.5.2 from 2011) with 3.16.7-ckt11-6-rc1 (HZ=100, compiled with gcc-4.9.2 from 2014).

For low system loads there is a 23% slowdown. Either this is the cost of checking the runqueue and updating statistics counters 1000 times per second instead of 100, or gcc-4.9.2 has much better optimizations than gcc-4.5.2.

In addition, for high system loads (two or more processes in the runqueue) the scheduling overhead is about 10% instead of 2%. Presumably this is the cost of performing 1000 context switches per second instead of 100.

Boot takes 32% longer, which makes sense because systemd tries to do plenty of things in parallel and this is the worst case for scheduling overhead.

The tests were performed with all unneeded services disabled, including spi0. On a normal system the extra scheduling overhead is less noticeable.

Again, this could be wrong because this is not an apples-to-apples comparison. I thought the potential performance improvements were worth mentioning anyway (either upgrade gcc or choose HZ somewhere between 100 and 1000). 

```
#!/bin/bash

echo "Boot time:"
journalctl -b  |  grep "Linux version\|lpj=\|Startup finished"

busywait() { n=0; while [ $n -lt 5000 ]; do n=$((n+1)); done }

echo "One task in runqueue:"
busywait &
time wait

echo "Two tasks in runqueue:"
busywait &
busywait &
time wait

echo "Three tasks in runqueue:"
busywait &
busywait &
busywait &
time wait
```

Result with 3.16.7-ckt11-5-ev3dev-ev3 (HZ=1000):
```
Boot time:
Jul 08 15:15:52 ev3dev kernel: Linux version 3.16.7-ckt11-5-ev3dev-ev3 (david@freyr) (gcc version 4.5.2 (Sourcery G++ Lite 2011.03-42) ) #1 PREEMPT Wed May 20 17:02:00 CDT 2015
Jul 08 15:15:52 ev3dev kernel: Calibrating delay loop... 144.12 BogoMIPS (lpj=72064)
Jul 08 15:16:33 ev3dev systemd[1]: Startup finished in 1.569s (kernel) + 44.368s (userspace) = 45.937s.
Jul 08 15:16:37 ev3dev systemd[429]: Startup finished in 577ms.
One task in runqueue:
real	0m6.469s
Two tasks in runqueue:
real	0m14.046s
Three tasks in runqueue:
real	0m21.556s
```

Result with 3.16.7-ckt11-6-rc1-ev3dev-ev3 (HZ=100):
```
Boot time:
Jul 08 15:30:42 ev3dev kernel: Linux version 3.16.7-ckt11-6-rc1-ev3dev-ev3 (pb@ev3dev) (gcc version 4.9.2 (Debian 4.9.2-10) ) #2 PREEMPT Wed Jul 8 14:15:01 UTC 2015
Jul 08 15:30:42 ev3dev kernel: Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
Jul 08 15:31:12 ev3dev systemd[1]: Startup finished in 1.578s (kernel) + 33.162s (userspace) = 34.741s.
Jul 08 15:31:16 ev3dev systemd[429]: Startup finished in 334ms.
One task in runqueue:
real	0m5.249s
Two tasks in runqueue:
real	0m10.707s
Three tasks in runqueue:
real	0m16.114s
```

dlech, 2015-07-08:
One again, thanks for the detailed info. I was actually just looking at this issue too.

Based on this feedback and https://github.com/ev3dev/ev3dev/issues/231#issuecomment-102194349, I am going to change HZ back to 100. I am also going to remove the lpj parameter from the kernel command line in case anyone want to play around with the HZ value in the future.

dlech, 2015-07-08:
These fixes have been released in kernel `3.16.7-ckt14-6-ev3dev-ev3` and `flash-kernel 3.35ev3dev5`

closed 2015-07-08 by dlech