ev3dev/ev3dev #338: rpi2 image vcgencmd fails "error while loading shared libraries"

BertLindeman, 2015-05-21:
Installed Raspberry pi2 image `rpi2-ev3dev-jessie-2015-05-19` (on a rPi2 :smile: ) Fiddled with it for some time, installed some packages etc. No sure if I installed something that caused this....

Symptom:
issued `sudo vcgencmd measure_temp` to see the temperature of the cpu/gpu.
Got: 
```
vcgencmd: error while loading shared libraries: libvcos.so: 
cannot open shared object file: No such file or directory
```
The file `libvcos.so` is part of package `libraspberrypi0`
at the moment it is already newest, but it *is* there: `/usr/lib/vc/libvcos.so`

My local circumvention:
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/vc`

Not sure if there is an official configuration for this.

dlech, 2015-05-22:
This comes from a raspberry pi foundation package that I rearranged, so it looks like I broke it. The upstream packages installs stuff in `/opt` which the debian packaging system complains about very much. So, I moved stuff into the proper directories. I'll have a look at it tomorrow.

BertLindeman, 2015-05-22:
Thanks, 
Not a hi priority thing. It's good to use the proper dirs.

dlech, 2015-05-23:
In `librasberrypi0` version `1.20150513-1ev3dev2`, I have moved the libraries to `/usr/lib/`. This fixes the need to use `LD_LIBRARY_PATH`. There are are some other issues with this that still go against Debian policy recommendations, but at least the command you gave works.

However, it would not surprise me if there are hard coded paths somewhere the look for stuff in `/opt/vc`. So, if something else is broken, open an new issue and we may end up moving the stuff back the way it is in upstream Raspbian from the RPi foundation.

BertLindeman, 2015-05-23:
On the pi2B:
apt-get update && apt-get upgrade
```
The following packages will be upgraded:
   libraspberrypi-bin (1.20150513-1ev3dev1 => 1.20150513-1ev3dev2)
   libraspberrypi0 (1.20150513-1ev3dev1 => 1.20150513-1ev3dev2)
   raspberrypi-bootloader (1.20150513-1ev3dev1 => 1.20150513-1ev3dev2)
```

Works:
```
echo "$LD_LIBRARY_PATH" ; vcgencmd measure_temp

temp=41.2'C
```

Same on the PI B+
Thanks again @dlech

closed 2015-05-23 by BertLindeman