ev3dev/ev3dev #339: "Platform not supported" on apt-get upgrade from ev3dev-rpi image rpi-ev3dev-jessie-2015-05-13

BertLindeman, 2015-05-23:
rPi-B+ came in :smile: Took rpi-ev3dev-jessie-2015-05-13.img and:
`time apt-get update && time apt-get -V upgrade` 

The upgrade does:
```
Calculating upgrade... Done
The following packages will be upgraded:
   fuse (2.9.3-15 => 2.9.3-15+deb8u1)
   libfuse2 (2.9.3-15 => 2.9.3-15+deb8u1)
2 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 192 kB of archives.
```
```
update-initramfs: deferring update (trigger activated)
Processing triggers for libc-bin (2.19-18) ...
Processing triggers for initramfs-tools (0.120) ...
update-initramfs: Generating /boot/initrd.img-3.18.13-5-ev3dev-rpi
Unsupported platform.
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Could not find a real solution for this. Might the packages come from the wrong repository?

Contents of `/etc/apt/source.list` not changed by me..
```
deb http://archive.raspbian.org/raspbian jessie main contrib non-free rpi

deb http://ev3dev.org/raspbian jessie main
```
On ev3dev-rpi2 no raspbian repository is used:
```
deb http://ftp.debian.org/debian jessie main contrib non-free

deb http://ev3dev.org/debian jessie main
```

A bit confused, now . . .

dlech, 2015-05-23:
I think it is `flash-kernel` that is causing the error. `flash-kernel` is what copies the kernel to the FAT partition. For RPi2, it is easy because currently there is only one model. For RPi1 though, there are many models, so I have probably missed some.

What is the output of `cat /proc/cpuinfo` and `cat /proc/device-tree/model`?

BertLindeman, 2015-05-23:
```
ev3dev-rpi:~# cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 2.00
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2708
Revision        : 0010
Serial          : 000000001b82f769
```

and 

```
@ev3dev-rpi:~# cat /proc/device-tree/model
Raspberry Pi Model B Plus
```

Thanks @dlech 
SOO much fun!
The EV3DEV image is a quick delivery vehicle to come to a good base system on RPi's.

dlech, 2015-05-23:
This should be fixed in `flash-kernel` version 3.35ev3dev4, uploading now...

BertLindeman, 2015-05-23:
Wow, that's FAST :clap: 

upgrade:
installing ` flash-kernel (3.35ev3dev2 => 3.35ev3dev4)`

Reset the holds (and fingers crossed..): 
```
echo "fuse install" | dpkg --set-selections
echo "libfuse2 install" | dpkg --set-selections
apt-get -V upgrade
```

Results in 
```
update-initramfs: deferring update (trigger activated)
Processing triggers for libc-bin (2.19-18) ...
Processing triggers for initramfs-tools (0.120) ...
update-initramfs: Generating /boot/initrd.img-3.18.13-5-ev3dev-rpi
Installing bcm2708-rpi-b-plus.dtb into /boot/dtb-3.18.13-5-ev3dev-rpi
Taking backup of dtb-3.18.13-5-ev3dev-rpi.
Installing new dtb-3.18.13-5-ev3dev-rpi.
Installing bcm2708-rpi-b-plus.dtb into /boot/dtb-3.18.13-5-ev3dev-rpi
Taking backup of dtb-3.18.13-5-ev3dev-rpi.
Installing new dtb-3.18.13-5-ev3dev-rpi.
flash-kernel: installing version 3.18.13-5-ev3dev-rpi
Version: Linux version 3.18.13-5-ev3dev-rpi (david@freyr) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #1 PREEMPT Wed May 13 12:35:10 CDT 2015
DT: y
Will use /dev/mmcblk0p1 as boot device.
Taking backup of kernel.img.
Installing new kernel.img.
Taking backup of bcm2708-rpi-b-plus.dtb.
Installing new bcm2708-rpi-b-plus.dtb.
```
ls of boot:
```
ll /boot
total 10728
-rw-r--r-- 1 root root  104551 May 13 19:32 config-3.18.13-5-ev3dev-rpi
lrwxrwxrwx 1 root root      24 May 23 17:49 dtb -> dtb-3.18.13-5-ev3dev-rpi
-rw-r--r-- 1 root root    4778 May 23 17:49 dtb-3.18.13-5-ev3dev-rpi
-rw-r--r-- 1 root root    4778 May 23 17:48 dtb-3.18.13-5-ev3dev-rpi.bak
drwxr-xr-x 3 root root   16384 May 23 17:49 flash/
-rw-r--r-- 1 root root 4951705 May 23 17:48 initrd.img-3.18.13-5-ev3dev-rpi
-rw------- 1 root root 1888221 May 13 19:40 System.map-3.18.13-5-ev3dev-rpi
-rw------- 1 root root 4002024 May 13 19:40 vmlinuz-3.18.13-5-ev3dev-rpi
```

OK let's reboot.
System did not voluntary come down by it's own (something with nfs or smbd).
But boot is quick and OK.
So solved. 
Thanks David.

closed 2015-05-23 by BertLindeman