ev3dev/ev3dev #340: Ubuntu 15.04 Internet Connection Sharing

AlienSM, 2015-05-26:
**Getting it to connect to the internet**
I have been struggling with this for a while. eventually got it working but have one small issue( Also thought it would be a good idea to describe what I did so others might be helped. )

I tried the GUI guide using [network manager](Shared to other computers) but it its written for 14.04 LTS as of 14.10 onwards there is no more option for "Method to Shared to other computers".  The interface of the network manger also changed.

I ended up followed the [old guide](https://github.com/ev3dev/ev3dev/wiki/Setting-Up-Linux-USB-Ethernet-Networking) and it mostly still works fine.

**Extra things I ended up doing to make sure it keeps on working after ubuntu reboots:**
[Made ip forwarding permanent on my ubuntu box](http://askubuntu.com/questions/311053/how-to-make-ip-forwarding-permanent)
[Saved my iptables on my ubuntu box so they get reloaded after a reboot/restart](http://askubuntu.com/questions/119393/how-to-save-rules-of-the-iptables)

**So after doing all this the one small issue is this:**
Ev3 connman service keeps on regenerating /etc/resolv.conf with "nameserver 127.0.0.1" so after a reboot you have to set the name server again to point to your router.  Is there away to stop it from regenerating the file every reboot or just to get generate it with the right value in?

***BIG PS HERE***
I'm fairly newish to playing with Linux so if there are better methods to doing these things please help and post.  That all said and done I'm a C++ Programmer by trade and have worked a bit with embedded Linux, done tons of 8bit embedded programming and electronics in the past.
So I can help with any technical tests or whatever would help and I'm not afraid of using the command line :) if anything I actually prefer it.  I also realize that getting it to work with just using the network manager in ubuntu 14.04 would be the golden ticket since it then makes it easier for new people to get it to work.  I will be more then willing to help with that also but I really don't know the network manager well enough.

dlech, 2015-05-26:
> Getting it to connect to the internet

This is a duplicate of #289. It look like it affects GNOME Shell as well, but *not* Unity. So, same workaround or patch from #289 can be applied. It would be nice if someone would add this to step 4 on [this page](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/?tabs-0=ubuntu-14-04). 

> Extra things I ended up doing to make sure it keeps on working after ubuntu reboots

Should not be needed since it is possible to use the NetworkManager method.

> So after doing all this the one small issue is this

In the future, please open a new issue for each separate item, no matter how small.

ConnMan provides it's own DNS server, which is why it always overwrites `/etc/resolv.conf`. If you like to use the command line, you can use the `connmanctl` command to configure the nameserver(s) for your connection. However, if you are using DHCP or the "Linux defaults" from brickman, these will be configured automatically for you.

> BIG PS HERE

We always welcome new contributors. (We need all the help we can get :wink:) Don't wait for someone to ask you to do something, just find something you are interested in and do it. Just be sure to let us know (by opening a new issue or commenting on an existing issue) so we aren't duplicating work. Don't be afraid to ask questions if you are not sure about something.

In addition to updating the Internet over USB tutorial, there are a number of other issues tagged as `HELP WANTED`. You can also test things out by just building a robot and letting us know how it goes. If you have any wisdom you learned from your background in electronics and embedded systems that you think could improve ev3dev, we'd like to hear about that too.

AlienSM, 2015-05-27:
> It would be nice if someone would add this to step 4 on this page.

Can I make a whole new tab for 15.04 with pictures?  Once I have figured out the hard way I will revert to Ubuntu's GUI and update the guide.

> In the future, please open a new issue for each separate item, no matter how small.

Sorry didn't mean to report two issues in one, it was intended only to get clarity on ConnMan.

> You can use the connmanctl command to configure the nameserver(s) for your connection.

I see the command is something like this if I understand it right?
`connmanctl configure <gadget service> --nameserver 192.168.x.x`
Will this persist between reboots?

> However, if you are using DHCP or the "Linux defaults" from brickman, these will be configured automatically for you.

How do I stop the automatic configuration?  I used Linux defaults can I just use custom settings and leave dns and gateway empty?

Is there a forum that I can join to ask stuff without having to create a possible duplicate issues?

MolarAmbiguity, 2015-05-27:
> Is there a forum that I can join to ask stuff without having to create a possible duplicate issues?
@AlienSM if you want to ask an informal question you can try the IRC. Its #ev3dev on freenode. More info is at http://www.ev3dev.org/support/#irc.

AlienSM, 2015-05-27:
>@AlienSM if you want to ask an informal question you can try the IRC. Its #ev3dev on freenode.

Thanks Molar

dlech, 2015-05-27:
> Can I make a whole new tab for 15.04 with pictures?

You can, but it will be all the same except for step 4.

> I see the command is something like this if I understand it right?

I think it is `config` rather than `configure` and `nameservers` (plural). You probably want to enable autoconnect as well.

    connmanctl services
    connmanctl config <gadget service> --nameservers 192.168.x.x --autoconnect

Running `connmanctl` by itself runs in interactive mode and has tab completion, which you may find easier.

> Will this persist between reboots?

Yes.

> How do I stop the automatic configuration?

If by automatic configuration, you mean DHCP, you need to specify a manual IP address.

> I used Linux defaults can I just use custom settings and leave dns and gateway empty?

The "Linux defaults" in brickman set the dns and gateway. However, if you manually configure the network connection yourself, you can of course leave the gateway and dns empty. Of course, you will not be able to access the Internet this way.

> Is there a forum that I can join to ask stuff without having to create a possible duplicate issues?

It's OK to just ask questions by opening a new GitHub issues. And you can search existing (open and closed) issues first to try to avoid duplicates.

AlienSM, 2015-05-27:
Got it all working with reboot and everything

Thanks for all the help.

>You can, but it will be all the same except for step 4.

From step 2 its a bit different to get to the network connections instead of settings if the patch isn't applied.  What I will do is see if I can make a nice option to show the different paths for unity and gnome and also make a note at step 4 that to get it in network settings you need to apply the patch.  Will think about it a bit and play around when I have reset all my network shenanigans to normal so that I can test the GUI and write it up nicely.  Once that is done I will apply the patch and then test that also.  Was thinking that possibly doing step 2 to step 4 in a sub tab box with tabs for "unity/gnome with patch" and "gnome without patch."

> Running connmanctl by itself runs in interactive mode and has tab completion, which you may find easier.
Thanks this was very help full and saved allot of time.

dlech, 2015-05-28:
I think it may be a bit much to add instructions about applying the patch. Although I suppose it is worth mentioning so that others can make some noise on the launchpad bug so that hopefully it gets fixed in Ubuntu.

AlienSM, 2015-05-28:
>Although I suppose it is worth mentioning so that others can make some noise on the launchpad bug so that hopefully it gets fixed in Ubuntu.

For what it is worth I have confirmed the bug and subscribed to it.  I see all these things increase the bug heat also.

>Improving tab interface on website #342

Thanks for that.

closed 2015-05-28 by AlienSM