ev3dev/ev3dev #341: Refering people to the ev3dev website

MolarAmbiguity, 2015-05-27:
I was trying to help out @AlienSM in #340. I linked to http://www.ev3dev.org/support/#irc. When I click on that link the (extremely helpful) navigation bar blocks the heading of the content in addition to the first paragraph. Whilst a minor nitpick this feels like an easy fix that could greatly improve user experience.

I'm using firefox 38.0 on Linux Mint. I'm curious to see if others can reproduce this. I don't know what would need to be edited to resolve this issue.

ddemidov, 2015-05-27:
Confirmed on chromium 43.0.2357.81 (Arch linux)

MolarAmbiguity, 2015-05-27:
Also confirmed in Midori 0.4.3 on Mint. I think its safe to assume this is universal.

dlech, 2015-05-27:
I thought it was just me...

dlech, 2015-05-27:
How does it look now?

closed 2015-05-27 by dlech

ddemidov, 2015-05-27:
Looks great in chromium!

MolarAmbiguity, 2015-05-28:
Looking good @dlech