ev3dev/ev3dev #342: Imroving tab interface on website

dlech, 2015-05-28:
This is a spin-off of some of the suggestions in <https://github.com/ev3dev/ev3dev/issues/340#issuecomment-106067406>. On some of the tutorial pages on the website, we have separate tabs for Android/Linux/Mac/Windows. It was a quick hack that I think can be improved. Since the site is a static site, it is currently using javascript to implement the tabs, which is fine unless anyone knows some css tricks that are better.

There are 2 main things I would like to accomplish...

- [ ] __Have all of the content in one markdown page.__ Currently, the content of each tab is in a separate markdown page, so clicking the "Edit on GitHub" link for those pages is not very useful. Also it is a pain to add new tabs and get the syntax right. Instead, I think we could use the [attribute list definitions](http://kramdown.gettalong.org/syntax.html#attribute-list-definitions) feature of kramdown to keep the markdown readable and maintainable but have everything on one page.
- [ ] __Use the HTML5 [localStorage](http://diveintohtml5.info/storage.html) feature to remember the tab selection between pages.__ If someone is a Windows user, they shouldn't have to click on the Windows tab on each page because it is the last one in the list. To do this, we need to standardize the tab names.

For tab name standardization, I propose the following:

1. A top level tab group called `os` that is operating systems with possible values of `Android`, `iOS`, `Linux`, `Mac OSX` and `Windows`
2. A secondary level tab group called `os-releaease` that will depend on the top level tab. That is, one of these values is stored for each top level tab. So, Android might have `4.x`, 5.x`; iOS `6`, `7`, `8`; Linux `Ubuntu 14.04`, `Ubuntu 14.04 (GNOME Shell)`, `Mint 17.1 (Cinnamon)`; Mac OSX `10.8`, `10.9`, `10.10`; Windows `7`, `8`, `10`. This gets a bit tricky though, especially with the Android and Linux releases.
3. A top level tab group called `programming-language` (or `lang` if we want something shorter). I don't see a need for subtabs in this group. The idea is that when listing programming examples, instead of just using shell (bash), we can have a tabs with as many programming languages as people want to submit.

The localStorage should only store the selected tab when it is clicked. If a page does not have a tab with the user's preferred selection, then a default tab should be shown, but that value should not be saved in localStorage. The default tab should be `Windows` for `os` and and `shell` for `lang`.

MolarAmbiguity, 2015-05-28:
I'm not sure how it works, but could we scrape the user's os from their browser?

dlech, 2015-05-28:
Ooh. That would be a better "default" for the `os` tab. And yes, I'm pretty sure you can do this with javascript.

MolarAmbiguity, 2015-05-28:
A quick search (http://stackoverflow.com/questions/9514179/how-to-find-the-operating-system-version-using-javascript) tells me we could use some variation of ```navigator```. You can test that out by putting it in your browser's javascript console

WasabiFan, 2015-05-28:
What you're looking for is a user agent parser -- from what I remember, this one's pretty good: [demo site](http://faisalman.github.io/ua-parser-js/).

closed 2015-06-01 by dlech