ev3dev/ev3dev #343: Lego nxt light sensor

Oxel40, 2015-05-29:
Hi!
I wonder if there are any code examples for using the light sensor in batch or Python?
I can't understand how you would be able to take readings from the light sensor in bach or Python.
Thanks for all answers!!!

ddemidov, 2015-05-29:
ev3dev/ev3dev-lang#59 seems to indicate the nxt light sensor does work at least with python. Does the following work for you?
```py
>>> from ev3dev import *
>>> s = light_sensor()
>>> s.connected
True
```

Oxel40, 2015-05-30:
No @ddemidov, it does not work. Is there a module named ev3dev that i can download that would be able to take readings from the light sensor?
```
>>> from ev3dev import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named ev3dev
```

MolarAmbiguity, 2015-05-30:
When you are doing ```from ev3dev import *``` are you sshed into the brick?

Oxel40, 2015-05-30:
Yes @MolerAmbiguity, I am sshed using putty

MolarAmbiguity, 2015-05-30:
Also which version of ev3dev are you using? And where did you install python from?

Oxel40, 2015-05-30:
I installed python by folowing this instruction https://github.com/topikachu/python-ev3.
I don't remember which verison the ev3dev is.

ddemidov, 2015-05-30:
> I installed python by folowing this instruction https://github.com/topikachu/python-ev3.

I thought you were talking about [python-ev3dev](https://github.com/ddemidov/ev3dev-lang-python). My example won't work for @topikachu's bindings. You should probably get more help if you [open an issue ](https://github.com/topikachu/python-ev3/issues/new) in python-ev3 repository.

Oxel40, 2015-05-30:
Ok, Thanks alot @ddemidov!

ddemidov, 2015-05-30:
And, by the way, you *will* be able to use NXT light sensor with [python-ev3dev](https://github.com/ddemidov/ev3dev-lang-python).

closed 2015-05-30 by Oxel40