ev3dev/ev3dev #345: Bluetooth

burf2000, 2015-05-31:
So, with big companies, they have messed up wifi.

I want to be able to control multiple EV3 via a android tablet using EV3DEV.

rhempel, 2015-05-31:
Simon, I thought you wanted to be able to use Bluetooth? Is wifi OK? Remember, you can use the micro-usb wifi dongles if you use ev3dev

burf2000, 2015-05-31:
I want to use Bluetooth between Android and Ev3 and then EV3 to other EV3 (BT or USB)

dlech, 2015-05-31:
It is certainly possible to do this. However, there are not any remote control libraries that I know of yet, so you would have to write your own.

burf2000, 2015-05-31:
So how would I go about doing this?

dlech, 2015-05-31:
Do you have ev3dev up and running?

burf2000, 2015-05-31:
Not currently I have standard and Lejos, I was trying to get direct commands to work, if your saying EV3Dev can work I start installing it

dlech, 2015-05-31:
If the leJOS remote control library will run on Android, that may be a better choice at this point in time, but if you would like to try out ev3dev, [start here](http://www.ev3dev.org/docs/getting-started/). There are instructions on how to get a network connection to Android via Bluetooth. There is also a tethering option that can be used to create EV3-to-EV3 networking connections. There are a number of libraries for ev3dev, so pick a programming language that you like and try one.

burf2000, 2015-05-31:
Correct me if I am wrong but cant I just run a webserver on each of the EV3's and read write that way?

dlech, 2015-05-31:
You sure can (assuming you have a network connection).

ddemidov, 2015-06-01:
> However, there are not any remote control libraries that I know of yet

[RPyC](https://rpyc.readthedocs.org/en/latest/) (Remote Python Call) library works very well with python bindings: https://github.com/ddemidov/ev3dev-lang-python#remote-operation. Its possible to reuse same python code for local/remote control of ev3 motors/sensors. I know that python is not the most popular language for android development, but google search suggests it is doable.

burf2000, 2015-06-01:
Thank you everyone, I let you know how I get on, be awesome for someone to do a tutorial on using a smart device to control multiple ev3

ameysutavani, 2016-04-10:
> RPyC (Remote Python Call) library works very well with python bindings:

Anyone tried making python rpc from android to ev3??
Is there any similar alternative like RPyC for c/c++ bindings as well?

I want to run object recognition in opencv(preferably in c/c++) on android and command actions to ev3 further on....any ideas of doing that? (i know this can be done through lejos's remote api, but want to keep the whole project in same programming language as much as possible).
Thanks in advance.

ameysutavani, 2016-04-21:
I made this capability matrix, add/edit as applicable
[ev3 sw capabilities.ods.zip](https://github.com/ev3dev/ev3dev/files/229126/ev3.sw.capabilities.ods.zip)

closed 2017-07-25 by dlech