ev3dev/ev3dev #346: Using ev3 input port 1 as serial port?

Sashkakuhta, 2015-06-04:
So, i need to use input port 1 as uart, but cant find any actual instructions about it. This https://github.com/ev3dev/ev3dev/wiki/Frequently-Asked-Questions instructions is way too outdated. I tried setting in1 mode as other-uart and shorted yellow and blue wire on wire plugged in port 1. Then i tried cat to ttyS0 and ttyS1, none of them replied with something when i was sending something. How do i do it then?

dlech, 2015-06-04:
If you want to use input port 1 to debug kernel messages, you need to edit `/etc/modprobe.d/ev3.conf` and `/etc/sysctl.d/ev3.conf`. This make the port unusable for anything else.

If you would like to keep the port configurable, but still use the UART function, you can change the port mode to `UART`. See <http://www.ev3dev.org/docs/ports/legoev3-input-port/> and <http://www.ev3dev.org/docs/drivers/lego-port-class/>.

dlech, 2015-06-04:
Oops, I meant `other-uart` instead of `UART` for the mode. Corrected above.

dlech, 2015-06-04:
Also, there are symlinks for the input port ttys so you don't have to sort out which one is which. There should be a `/dev/tty_in1`.

Sashkakuhta, 2015-06-04:
Its actually called tty_ev3:in1. What i want is connecting to another device via uart from ev3. I tried to set mode to other-uart and ev3-uart but ttys are silent even when connecting rx and tx

dlech, 2015-06-04:
Sounds like you need to do `sudo apt-get update; sudo apt-get upgrade`. It is not supposed to have the `ev3:` in the link (this causes UART sensors to not work). It won't affect what you are trying to to though.

Maybe you have your rx and tx crossed. It is working for me.

Sashkakuhta, 2015-06-04:
I updated it, messed with minicom a bit and it started working, thanks!
Minicom needs to be launched withs -s flag and setted to tty_in1 port, then it should just start working.

closed 2015-06-04 by Sashkakuhta