ev3dev/ev3dev #348: Open Roberta and ev3dev

dlech, 2015-06-10:
I received a nice email from some folks with [Open Roberta](http://www.open-roberta.org/en/willkommen/) last week:

> I would like to suggest our project Open Roberta to you.

>The programming platform Open Roberta Lab is a completely open source development. Both the software and the open source developer tools are available via Fraunhofer servers. In a first step, the development team at the Fraunhofer Institute for Intelligent Analysis and Information Systems reached out to teachers, IT and education experts within the Roberta-network as well as to Universities and their students to involve them in the development work. In the second step, the open-source community has been opened to all interested programmers and programmers. Following the main idea of the Roberta project, this should especially addresses female students and encourages them to become role models for the next generation of students.

> So far we have integrated the LEGO Mindstorms EV3 in the lab. To generate the code we are using the lejos libraries, because they offer us full and easy support. But we are also interested in other systems like ev3dev, to increase the performance in general.

> Maybe you are interested in our project. I like to invite you to join our forum  https://groups.google.com/forum/#!forum/open-roberta. Please feel free to share this mail with your developers.

They have a very nice looking programming environment that is similar to Scratch and Snap!. If anyone is interested in trying to make it work with ev3dev, by all means, go for it. In particular, I thought this might be of interest to @sharived.

rhempel, 2015-06-11:
Sebastian Trella is a member of the MCP15 with me - if you manage to make some of their apps work with ev3dev, that would be fantastic.

sharived, 2015-06-11:
I was trying to bring up RoboSnap on OpenRoberta, but haven't had much time to try it out.  Will try first on Lejos and if I can figure out how to make it work on ev3dev, will let you know.

ensonic, 2015-06-25:
I am also interested here. This needs quite a few changes on the openroberta side right now, since they generate java code for lejos and as far as I know we don#t have the apis for that and the openroberta project also suffers from the long startup time (when they send a new jar).

I think one part would be to suggest the nicest runtime they should target if they rework their code generator.

ensonic, 2015-07-03:
I have a quick hack of the ev3menu part that talks to openroberta writting in python for the ev3. I am discussion the json protocol they use for openroberta on their dev list already. Also right now I get a lejos-specifc java-jar. Next week I'll study the code-generator part and work on a proposal/proto to configure the programming language, so that we could e.g. get python for ev3dev.

ensonic, 2015-07-24:
I have submitted all the code allowing an initial end-to-end demo. See https://mp-devel.iais.fraunhofer.de/wiki/display/ORDevel/Open+Roberta+on+ev3dev for a brief how-to. This is by no means complete. Especially the python code-generator is not feature complete.

rhempel, 2015-07-24:
Stefan - this looks really cool - I'm assuming that you are already in contact with Sebastian Trella on this, correct?
I am slowly ramping up on Python - I was using Lua for the longest time because it is very fast and has an extremely small memory footprint, which was great when the NXT had only 64 K of RAM. Sow that we have multi megabytes, Python makes a lot more sense, and it's got so many libraries built in and ready to use, including the test lib.
I'm going to be away from keyboard for the next 2.5 weeks, but I'm clearing my desk to dive into the Python side of ev3dev.
Keep up the great work, and let me know if you need an introduction to Sebastian.

ensonic, 2015-07-30:
Some more throughts: what I have currently in the openroberta-codebase is a python connector that one need to launch from the 'file brower'. This will start the session with the cloud ide and then receive and run code. I don't like this a lot since it is a launcher on top of a launcher. What about this long term plan:

I create a robertalab.service. This would be a head-less deamon with a dbus iface. It can be asked to connect to the cloud-ide and once connected receive and run programs. 

We implement ev3dev/ev3dev/#129
I would then implement a roberta-lab app, that will open a dialog, talk to the robertalab.service to start a session (or terminate the exiting one). Closing the app will only close the dialog.

We implement the systray spec
http://standards.freedesktop.org/systemtray-spec/systemtray-spec-0.2.html
or something simpler inspired by it so that my service can tell brickman to show a status icon

dlech, 2015-07-30:
I think this sounds like a reasonable plan. Also, I don't have a problem with putting openroberta stuff directly in brickman. So, for example if you just want to build an openroberta status icon into brickman instead of implementing a system tray service, that would be OK.

dlech, 2015-10-23:
I would say that is is no longer in the planning stages. :wink:

A big shout out to @ensonic for moving forward with this and making it work.

closed 2015-10-23 by dlech

rhempel, 2015-10-23:
Yes, big thanks to @ensonic for making this happen! I am getting my arms around pycharm this weekend and hopefully improving (without breaking) the pure Python binding. Thanks to @ddemdov and @antonvh for helping out and again to @ensonic for being patient through the breaking changes.