ev3dev/ev3dev #349: Running a program on ev3dev

DarthTater, 2015-06-20:
I have no idea how to run a file on  ev3dev, i am interested in programing with node.js, but i cant get files to show up on the ev3 brick ='(
help plz...

dlech, 2015-06-20:
There are a number of ways to run a program.

To get the program to show up on the EV3 brick, you need to create a user account as described in the [Getting Started](http://www.ev3dev.org/docs/getting-started/#set-up-a-new-user) guide. Then any files in your home folder will be visible in the file browser on the EV3 brick.

You can also run programs directly from a remote terminal via SSH. If you want to draw on the screen, this complicates things a bit. See #322. I've made a [script](https://gist.github.com/dlech/4c9b268571af59cb7bb4) to simplify this a bit.

WasabiFan, 2015-06-20:
@dlech How does the file browser determine what is actually launched? If one selects a .js file in the file browser, will it be able to find a file association for the Node binary and run that? Or will it just give up of you try to run a file which isn't a standard executable (shell script or binary)?

ddemidov, 2015-06-20:
The standard unix practice of shebanging a js file and making it executable works for me:
```
$ cat hello.js 
#!/usr/bin/node
console.log("Hello World!");
$ chmod +x ./hello.js
$ ./hello.js
Hello World!
```

WasabiFan, 2015-06-20:
That's what I was worried about... So if @DarthTater wants to run a JS file it'll require either adding the execution info to the file (as @ddemidov showed) or running it from an ssh session with "nodejs foo.js".

I should add this info to the readme for the JS repo.

dlech, 2015-06-20:
Right, only executable files can be run from brickman (`chmod +x`). And for any executable text file, it needs the shebang to tell it what should actually be used, whether it is `node` or `python` or `bash` or whatever.

DarthTater, 2015-06-21:
How do I set up the user, there is the code on the getting started page, but idk where to enter it

dlech, 2015-06-21:
You need to follow the entire Getting Started guide from beginning to end. In step 4, you will setup a network connection and open a remote terminal using SSH (PuTTY on Windows). That is where you enter the "code".

DarthTater, 2015-06-22:
what is the file extension for a ev3dev executable?

MolarAmbiguity, 2015-06-22:
ev3dev is linux, you can run pretty much anything. There is no single ev3dev executable.

DarthTater, 2015-06-22:
what type of file will the ev3 recognise as an executable and run

MolarAmbiguity, 2015-06-22:
ev3dev will recognise anything that you chmod -x. For node, look at @ddemidov's answer higher up in this post.

rhempel, 2015-06-22:
You mean `chmod +x` of course, and for scripting languages, you need to add the correct shebang line to the beginning of the script.
@DarthTater, have you done any programming at all on a command line or written scripts that run on Windows, Linux or OSX? - it's OK if ou haven't - we're here to help, but we need to have some idea of your level of expertise to give you meaningful answers.

MolarAmbiguity, 2015-06-22:
@rhempel whoops, I've updated my replies with ```chmod +x```. I don't know what I was thinking.

closed 2015-07-30 by dlech