ev3dev/ev3dev #350: Building ev3devKit

ensonic, 2015-06-26:
I've send a PR for some easy README updates, but there are a few more things. 

* I also had to bump vala from the vala-team PPA (https://launchpad.net/~vala-team/+archive/ubuntu/ppa). I can mention this in the README too.

* there is no "libgrx-dev" (on ubuntu trusty). Since there is a version in this repo, should that be used instead? If I build the desktop version, should I build the X11 version of grx?

dlech, 2015-06-26:
> I also had to bump vala from the vala-team PPA

That works. You can also install vala 0.24 from utopic, but the vala-team ppa might be easier.

> there is no "libgrx-dev"

This is an ev3dev package that does not exist in the greater debian/ubuntu world. You need to install it from the ev3dev trusty archive. `deb http://ev3dev.org/debian trusty main`.

> Since there is a version in this repo, should that be used instead?

Nope, use the packaged version unless you need to make changes to libgrx.

> If I build the desktop version, should I build the X11 version of grx?

Nope, There is  some other trickery going on. We don't use the X11 version, even for "desktop".

---

The readme from [brickman](https://github.com/ev3dev/brickman) should also help. It is setup essentially the same a ev3devKit.

ensonic, 2015-07-01:
I've updated the PR. git clone --recursive works great and is already available in trusty which you suggest as a development base. So this procedure could also be recommended in the [brickman README](https://github.com/ev3dev/brickman).

ensonic, 2015-07-01:
Hmm, now my build fails at:

```
> make
[  2%] Built target fix-vapi
[ 19%] Built target ev3devkit_data
[ 75%] Built target ev3devKit
[ 83%] Built target ev3devKit-desktop
[ 90%] Built target ev3devKit-desktop-typelib
[ 91%] Generating demo/devices_demo.c
error: Package `ev3devKit-desktop' not found in specified Vala API directories or GObject-Introspection GIR directories
Compilation failed: 1 error(s), 0 warning(s)
# ...
> find . -name "*desktop*vapi*"
./ev3devkit-desktop-0.1.vapi
```

Any idea?

dlech, 2015-07-01:
I think I have a build dependency missing somewhere that causes things to be built out of order sometimes. If this is what I am thinking of, just running `make` again will complete the build.

Also, what are you using for your `cmake` command?

ensonic, 2015-07-02:
cmake ../ev3devKit -DEV3DEVKIT_DESKTOP=bool:Yes -DCMAKE_BUILD_TYPE=string:Debug -DBUILD_SHARED_LIBS=bool:Yes

I tried running make multiple times already. Unfortunately I have zero experience modifying a cmake project. I'll see if I can find something.

ensonic, 2015-07-02:
Voila, this fixes the build for me:
```
ln -s ./ev3devkit-desktop-0.1.vapi ./ev3devKit-desktop.vapi

ll *.vapi
-rw-r----- 1 ensonic eng 102097 Jun 26 15:37 ev3devkit-0.1.buggy.vapi
-rw-r----- 1 ensonic eng 102113 Jun 26 15:37 ev3devkit-0.1.vapi
-rw-r----- 1 ensonic eng   2932 Jul  1 17:43 ev3devkit-desktop-0.1.vapi
lrwxrwxrwx 1 ensonic eng     28 Jul  2 11:32 ev3devKit-desktop.vapi -> ./ev3devkit-desktop-0.1.vapi
```

I have a patch where I remove LIBRARY_NAME and DESKTOP_LIBRARY_NAME and instead use EV3DEVKIT_PACKAGE and EV3DEVKIT_DESKTOP_PACKAGE. Will send a new PR shortly.

dlech, 2015-07-03:
I just pushed a change that fixes the build for me. Is there something else I am missing?

ensonic, 2015-07-03:
Fixed. Thanks!

closed 2015-07-03 by ensonic