ev3dev/ev3dev #351: ev3dev Memory Usage

rhempel, 2015-06-29:
It seems that the latest distribution `linux-image-3.16.7-ckt11-6-rc1` is much slower to boot, and uses almost all available RAM.

The brick would hang after copying 7-10% during a `wget` of a kernel image over wifi, but that may have been an issue with the wifi dongle (ASUS A10N).

dlech, 2015-06-30:
I think some people might take your first sentence to mean that the kernel itself is using most of the ram, but I understand that you mean that the default settings of the image cause most of the RAM to be used by a number of different programs.

Here are some things I think are contributing to this:

*   I've added several new services in the 2015-05-01 image to make networking a bit more user friendly, namely avahi and nmbd. These let you connect by host name without the need for a DNS server or editing hosts files. avahi is needed for Mac and Linux and nmbd is needed for Windows. avahi is not too bad, but nmbd is a real memory hog.

    As mentioned in #316, I would like to modify the `raspi-config` program from Raspberry Pi for ev3dev so that we can easily turn these services on and off.

*   In the `3.16.7-ckt11-5` kernel, I have change `CONFIG_HZ` from 100 to 1000 to see what kind of difference it makes. This is discussed some in #337. I have not had any positive feedback on this. The only difference I have noticed is that it increased the spi0 CPU usage. We may want to change this back to 100. The additional CPU usage, I am sure causes a longer boot time as well.

*   I image some work could be done with our own brickman program as well to reduce memory usage.

rhempel, 2015-06-30:
Yeah, I'm OK with killing `nmbd` - or at least being able to configure fairly easily which services should be started.

Is `CONFIG_HZ` settable to some compromise between 100 and 1000?

It might be a good idea to provide a guideline for people to let them decide which services they can do without - we've only got 64 MB to work with :-) I can remember the first very lean distribution had LOTS of RAM left over after a boot.

And then we can pray to the overlords of EOL for components (especially the 64MB RAM device) to give us 128 or 256 MB to work with

dlech, 2015-06-30:
> Yeah, I'm OK with killing nmbd

So disable it by default? For ease of use, my thought was to enable it by default, but make it easy to turn off. Alternately, we could disable it by default and make it easy to turn on. If we go the second route though, we should make it so that you can enable it from brickman because the whole point is so you don't have to figure out the IP address before your connected with SSH.

> Is CONFIG_HZ settable to some compromise between 100 and 1000?

I imagine that you can set it to anything in between. These are just the two most common values. 100 may be just fine for our purposes though.

dlech, 2015-06-30:
I have also noticed that each SSH login takes a significant amount of memory. I've looked into using dropbear instead of openssh, but there are a number of issues/limitations/differences with dropbear that we would need to overcome before doing this. It is not a trivial swap.

rhempel, 2015-06-30:
The Hiawatha web server uses PolarSSL (now mbed TLS) for its SSL layer, not the ssh libraries. What we're waiting for is an ssh client that uses PolarSSL - but that by itself may not drop the RAM requirements for an SSH login.

For `nmbd` I think we can live with enabled by default - on the other hand, Brickman goes to the trouble of showing you the connected IP address, so what's the actual benefit of having `nmbd` or `avahi` enabled by default?

dlech, 2015-07-01:
> on the other hand, Brickman goes to the trouble of showing you the connected IP address

IP addresses can change. If you want to save your connection in PuTTY, using host name would be better. We could change brickman to show this hostname instead of the IP address.

rhempel, 2015-07-01:
For WiFi connections, I would recommend that people learn to set up their routers to give out fixed IP addresses based on MAC addresses. Or buy a cheap wifi router on clearance or at a grage sale so that you can dedicate it to EV3 - useful if you ever take your robot somewhere else for a demo.
Or learn to use the fixed USB addressing and get a hub if you have multiple EV3s.

G33kDude, 2015-07-01:
I think that the hostname should be configurable through brickman, which would make it easy to change before connecting to your network for the first time (which would be useful if you're setting up multiple EV3s. Hostname collision is confusing)

rhempel, 2015-07-01:
I'm not so sure we want to make `brickman` the all-singing all-dancing configuration place.
For example, I like that the USB config can be set to load `cdc` or `rndis` driver, and that we can set the IP to the default for Windows, Mac, or Linux.
Once that's done, I think we should focus on showing users how to set up their networking "properly" from the command line to values that do not overlap the defaults.
That way, if/when something goes wrong, they can _always_ go back to the USB cable method to fix things.

rhempel, 2015-07-01:
Here's part of the results of `ps aux` on my system:
```
root       269  0.0  2.5   7832  1504 ?        Ss   16:17   0:00 /usr/sbin/sshd -D
root       273  0.2  5.3   7844  3168 ?        Ss   16:17   0:02 /usr/sbin/connmand -n
root       274  0.0  1.5   4728   948 ?        Ss   16:17   0:00 /usr/lib/bluetooth/bluetoothd
root       333  0.2  4.9   6824  2940 ?        Ss   16:17   0:02 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
ntp        335  0.1  5.1   5816  3036 ?        Ss   16:17   0:00 /usr/sbin/ntpd -p /var/run/ntpd.pid -g -u 100:103
root       344  3.3 11.3  54844  6752 tty1     Rsl+ 16:17   0:28 /usr/sbin/brickman
www-data   345  0.0  1.3  19248   832 ?        Ssl  16:17   0:00 /usr/sbin/hiawatha
root       452  0.4  0.0      0     0 ?        S    16:17   0:03 [RTW_CMD_THREAD]
root       456  0.0  0.0      0     0 ?        S<   16:17   0:00 [cfg80211]
root       496  0.0  0.0      0     0 ?        S<   16:18   0:00 [kworker/0:1H]
root       508  0.0 11.5  27384  6876 ?        Ss   16:19   0:00 /usr/sbin/winbindd
root       509  0.0  7.8  22556  4668 ?        Ss   16:19   0:00 /usr/sbin/nmbd -D
root       510  0.0 12.5  27512  7436 ?        S    16:19   0:00 /usr/sbin/winbindd
```
Once I'm all configured the first time, `brickman` can be an optional load - especially if I have some way of automatically loading a script.
Not sure why we have two instances of `winbindd` running - and yes, that coupled with `nmbd` blows away about 30% of available RAM!

dlech, 2015-07-02:
USB and Bluetooth IP addresses are DHCP when using Internet Connection sharing unless you go to the trouble of setting up a static address.

So, what I am hearing from @rhempel is that we should set the bar at the level "you *must* learn networking before your can use ev3dev". However the feedback I have received from most ev3dev users (that have taken the time to give feedback) is "Are we setting the bar too high? Wouldn't it be nice if you didn't have to learn networking (and some Linux) before you can even to the most basic things with ev3dev?"

I do agree that I only want minimal configuration in brickman, however, I think hostname might be one of these things. It is really only an issue when you have multiple EV3s though.

dlech, 2015-07-02:
`winbindd` should not be running at all. I'm pretty sure it is disabled by default in the 2015-05-20 image.

rhempel, 2015-07-02:
@dlech - I'm using an image I generated around that time - but I've done `apt-get update` and `apt-get upgrade` from our official repos (not my local ones) since then. I think I need to go over which services are enabled at startup.

As far as setting the bar at a certain level for networking, maybe I am not communicating well. Brickman is a great home screen for initial boot up - it allows you to set up defaults for WiFi, USB, and Bluetooth and makes it clear what the connected IP address is as well as battery levle.

Once that's done users will need to learn a little bit about networking to customize things if they want to be able to share directories or communicate with multiple EV3 bricks.

We need to be careful to separate "I don't want to learn Linux" from "How do I get Python files to my brick" from "I need an IDE to write code" users. In most cases, having a friend get them over the hump of the initial configuration is a big help.

Many users have trouble with the idea that the EV3 is just another computer like the one in their router. Which is part of the reason I wanted to put Hiawatha on the brick. Most routers have a "set everything to defaults" mode, then you connect to it and open up a  web page where all the real settings can get done with a web interface. That's where I see this eventually going...but of course, I'm open for discussion and comments.

G33kDude, 2015-07-02:
I think that by default there should be a samba share running on the brick.
That way, as soon as the brick is set up with Wi-Fi you can drop your
programs on there without any third party program. This would jumpstart the
initial setup, since as soon as you're on the network you will have super
simple file system access. It would also solve the "how do I get python
files to my brick" problem.

rhempel, 2015-07-02:
Another alternative is a CGI script that lets you upload files over any IP connection :-)

G33kDude, 2015-07-02:
Although that could be more flexible, it would require you to manually copy files over every time you modify them, wouldn't it? Also, samba shares are supported natively in Windows, Macs, and Ubuntu (not sure about other distros), (ChromeOS too with an extension) so I'm not sure for which devices this would work better than samba. However, it'd be interesting nonetheless, especially if there was an integrated editor like in OwnCloud.

I think we're getting away from the issue topic, though

rhempel, 2015-07-02:
No, you're right on - I am using smbnetfs to mount remote Samba shares on the EV3, it could work the other way around - need to see what the smbserver footprint is.

pabr, 2015-07-06:
I can confirm that ev3-ev3dev-jessie-2015-05-20 feels more sluggish (as experienced in a shell over ssh over usbnet/cdc) than the version I was using 6 months ago. Disabling legoev3_analog.ko (and therefore spi0) solves the problem.
 
I am not sure why CONFIG_HZ=1000 increases spi0 CPU usage. Maybe it causes so much overhead that the spi0 tasklet ends up consuming a larger portion of what little CPU time is left. I will recompile with HZ=100 and report here if that helps figure out anything. Regardless, I don't think HZ=1000 was ever intended for embedded systems running at 300 MHz.

JorgePe, 2015-07-07:
Just my 2 cents...
- I'm just an average linux user but I feel that those who want to jump to ev3dev should be aware that a minimal understanding of linux is important; I'm convinced that all who jumped in had already some previous background. And nowadays linux is no more a monster as the Raspberry Pi has been showing.
- I would prefer if ev3dev stays as minimalist as possible. I'm not a good programmer so I abuse from Google and install lot of dependencies that sometimes is a miracle when some code runs OK with such few RAM. Some things like samba shares or network management could be left optional, off by default.
- the idea of using CONFIG_HZ=1000 is for real time applications? I'm also not sure if the EV3 ARM handles, doesn't that generates much more interrupts?

dlech, 2015-07-07:
> the idea of using CONFIG_HZ=1000 is for real time applications?

You can't really call it realtime like the realtime patch set, but it may give better results in some cases. 

> I'm also not sure if the EV3 ARM handles, doesn't that generates much more interrupts?

Not more interrupts, but more CPU wakeups to handle non-interrupt things.

dlech, 2015-09-07:
Closing since we've fixed the main issue here, which was the sluggishness introduced by `HZ=1000`. There were a few other discussions that spawned off of this topic, so if there are any actionable items or items that need further discussion, please open a new issue and reference the appropriate comments here.

closed 2015-09-07 by dlech