ev3dev/ev3dev #352: Failed to read ACL on /var/log/journal

rhempel, 2015-07-01:
After boot (Linux version 3.16.7-ckt11-6-rc1-ev3dev-ev3) , I am seeing messages like
```
[  130.758202] systemd-journald[63]: Failed to read ACL on /var/log/journal/c7e7e3e1f0d7a493a5215b715556b154/user-1000.journal, ignoring: Operation not supported
```
in the log - does this signal an issue with my SD card or some other problem?

dlech, 2015-07-01:
We don't have ACLs enabled in the kernel. I could find very little information on the subject, but in what little info I found, it suggested that ACLs reduce performance hit on disk I/O. I couldn't find any hard data to back this up though, so I don't know if it is true and if it is, how much.

ACLs are enabled in the Raspberry Pi kernel and it runs off of and SD card too. I'm thinking I'm just going to enabled them in the next kernel release unless someone wants to do some performance tests first.

dlech, 2015-07-08:
Duplicate of #257.

closed 2015-07-08 by dlech