ev3dev/ev3dev #353: TPLink 725 WiFi device no longer detected

rhempel, 2015-07-01:
My TP-Link 725 is no longer detected - `dmesg` and `lsusb` shows:

```
175.712739] usb 1-1: new full-speed USB device number 2 using ohci
rhempel@ev3dev:~$ lsusb
Bus 001 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

I have the firmware file, I suspect that the PID/VID pair is no longer handled properly - I'll check the source later this afternoon.

rhempel, 2015-07-01:
Doh, I just needed to set `CONFIG_R8188EU=m` in my local kernel config.
@dlech we probably want the default kernel config to have this. Your private `.config` has it because the kernels you release do have support for this module.
I'll close when this is done so we don't lose sight of it.

dlech, 2015-07-01:
I don't use a private `.config` for releases. The configuration is always generated from the default configuration.

I think the difference is that with the release, we include an out of kernel module package for `rtl8188cu`. If the mainline kernel driver for this chip is good enough now, we can use it instead of an external module.

dlech, 2015-07-08:
We are now using `CONFIG_R8188EU=m` in kernel `3.16.7-ckt14-6-ev3dev-ev3` and have dropped the out of tree module (separate package).

closed 2015-07-08 by dlech