ev3dev/ev3dev #354: stuck at booting after running apt-get upgrade

ensonic, 2015-07-02:
I am using ev3-ev3dev-jessie-2015-05-20 and run apt-get update + apt-get upgrade today. After the reboot it boots, but gets stuck at some point before launching brickman. Neither wifi nor usb-ethernet are up at that point. Since the default bootscreen is not telling any boot progress I have not idea where it died.

```
 apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  dbus ev3-systemd fuse libdbus-1-3 libfuse2 libgnutls-deb0-28 libgnutls-openssl27 libgudev-1.0-0 libpam-systemd libruby2.1 libsqlite3-0 libsystemd0 libtasn1-6 libudev1 linux-libc-dev perl perl-base perl-modules ruby ruby2.1 systemd systemd-sysv tzdata udev
  wpasupplicant
25 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 16.1 MB of archives.
After this operation, 581 kB disk space will be freed.

Do you want to continue? [Y/n] y
Get:1 http://ftp.debian.org/debian/ jessie/main perl armel 5.20.2-3+deb8u1 [2,072 kB]
Get:2 http://ev3dev.org/debian/ jessie/main ev3-systemd armel 3 [5,058 B]   
Get:3 http://ftp.debian.org/debian/ jessie/main perl-base armel 5.20.2-3+deb8u1 [1,100 kB]                                                                                                                                                                                   
Get:4 http://ftp.debian.org/debian/ jessie/main perl-modules all 5.20.2-3+deb8u1 [2,551 kB]                                                                                                                                                                                  
Get:5 http://ftp.debian.org/debian/ jessie/main libudev1 armel 215-17+deb8u1 [51.1 kB]                                                                                                                                                                                       
Get:6 http://ftp.debian.org/debian/ jessie/main udev armel 215-17+deb8u1 [850 kB]                                                                                                                                                                                            
Get:7 http://ftp.debian.org/debian/ jessie/main libsystemd0 armel 215-17+deb8u1 [83.6 kB]                                                                                                                                                                                    
Get:8 http://ftp.debian.org/debian/ jessie/main systemd armel 215-17+deb8u1 [2,220 kB]                                                                                                                                                                                       
Get:9 http://ftp.debian.org/debian/ jessie/main systemd-sysv armel 215-17+deb8u1 [34.2 kB]                                                                                                                                                                                   
Get:10 http://ftp.debian.org/debian/ jessie/main libpam-systemd armel 215-17+deb8u1 [112 kB]                                                                                                                                                                                 
Get:11 http://ftp.debian.org/debian/ jessie/main libdbus-1-3 armel 1.8.18-0+deb8u1 [151 kB]                                                                                                                                                                                  
Get:12 http://ftp.debian.org/debian/ jessie/main dbus armel 1.8.18-0+deb8u1 [250 kB]                                                                                                                                                                                         
Get:13 http://ftp.debian.org/debian/ jessie/main libgnutls-openssl27 armel 3.3.8-6+deb8u1 [138 kB]                                                                                                                                                                           
Get:14 http://ftp.debian.org/debian/ jessie/main libtasn1-6 armel 4.2-3+deb8u1 [43.1 kB]                                                                                                                                                                                     
Get:15 http://ftp.debian.org/debian/ jessie/main libgnutls-deb0-28 armel 3.3.8-6+deb8u1 [624 kB]                                                                                                                                                                             
Get:16 http://ftp.debian.org/debian/ jessie/main libsqlite3-0 armel 3.8.7.1-1+deb8u1 [378 kB]                                                                                                                                                                                
Get:17 http://ftp.debian.org/debian/ jessie/main fuse armel 2.9.3-15+deb8u1 [68.7 kB]                                                                                                                                                                                        
Get:18 http://ftp.debian.org/debian/ jessie/main libfuse2 armel 2.9.3-15+deb8u1 [124 kB]                                                                                                                                                                                     
Get:19 http://ftp.debian.org/debian/ jessie/main libgudev-1.0-0 armel 215-17+deb8u1 [38.3 kB]                                                                                                                                                                                
Get:20 http://ftp.debian.org/debian/ jessie/main tzdata all 2015d-0+deb8u1 [192 kB]                                                                                                                                                                                          
Get:21 http://ftp.debian.org/debian/ jessie/main linux-libc-dev armel 3.16.7-ckt11-1 [976 kB]                                                                                                                                                                                
Get:22 http://ftp.debian.org/debian/ jessie/main libruby2.1 armel 2.1.5-2+deb8u1 [3,027 kB]                                                                                                                                                                                  
Get:23 http://ftp.debian.org/debian/ jessie/main ruby2.1 armel 2.1.5-2+deb8u1 [276 kB]                                                                                                                                                                                       
Get:24 http://ftp.debian.org/debian/ jessie/main ruby all 1:2.1.5+deb8u1 [9,620 B]                                                                                                                                                                                           
Get:25 http://ftp.debian.org/debian/ jessie/main wpasupplicant armel 2.3-1+deb8u1 [769 kB]                                                                                                                                                                                   
Fetched 16.1 MB in 1min 1s (264 kB/s)                                                                                                                                                                                                                                        
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 28887 files and directories currently installed.)
Preparing to unpack .../perl_5.20.2-3+deb8u1_armel.deb ...
Unpacking perl (5.20.2-3+deb8u1) over (5.20.2-3) ...
Preparing to unpack .../perl-base_5.20.2-3+deb8u1_armel.deb ...
Unpacking perl-base (5.20.2-3+deb8u1) over (5.20.2-3) ...
Setting up perl-base (5.20.2-3+deb8u1) ...
(Reading database ... 28887 files and directories currently installed.)
Preparing to unpack .../perl-modules_5.20.2-3+deb8u1_all.deb ...
Unpacking perl-modules (5.20.2-3+deb8u1) over (5.20.2-3) ...
Preparing to unpack .../libudev1_215-17+deb8u1_armel.deb ...
Unpacking libudev1:armel (215-17+deb8u1) over (215-17) ...
Setting up libudev1:armel (215-17+deb8u1) ...
Processing triggers for libc-bin (2.19-18) ...
[... cut ...]
update-initramfs: Generating /boot/initrd.img-3.16.7-ckt11-5-ev3dev-ev3
flash-kernel: installing version 3.16.7-ckt11-5-ev3dev-ev3
Generating kernel u-boot image... done.
Will use /dev/mmcblk0p1 as boot device.
Taking backup of uImage.
Installing new uImage.
find: `/etc/flash-kernel/ubootenv.d': No such file or directory
Generating boot script u-boot image... done.
Taking backup of boot.scr.
Installing new boot.scr.
root@ev3dev:~# Connection to ev3dev.local closed by remote host.
Connection to ev3dev.local closed.
```

ensonic, 2015-07-02:
Is the port-1 serial console enabled by default? I am not getting anything to show up there?

I also booted into a new image and tried to look at the journal from the memory card that gets stuck when loading. The initial part is identical to the journal of a new log. The journal where booting gets stuck is online at https://gist.github.com/ensonic/60ae56e874b19c2fd2ef. The log on a system that keeps booting continues as
```
Jan 01 01:29:19 ev3dev dbus[271]: [system] Rejected send message, 2 matched rules; type="method_return", sender=":1.4" (uid=0 pid=306 comm="/usr/sbin/connmand -n ") interface="(unset)" member="(unset)" error name="(unset)" requested_reply="0" destination=":1.7" (uid=0 pid=464 comm="/usr/sbin/brickman ")
Jan 01 01:29:19 ev3dev connmand[306]: ntp: time slew +488551904.074243 s
Jun 25 15:21:03 ev3dev connmand[306]: usb0 {add} route 10.42.0.0 gw 0.0.0.0 scope 253 <LINK>
Jun 25 15:21:03 ev3dev connmand[306]: usb0 {add} route 10.42.0.1 gw 0.0.0.0 scope 253 <LINK>
Jun 25 15:21:03 ev3dev connmand[306]: usb0 {add} route 0.0.0.0 gw 10.42.0.1 scope 0 <UNIVERSE>
Jun 25 15:21:03 ev3dev connmand[306]: usb0 {add} route 87.106.208.187 gw 10.42.0.1 scope 0 <UNIVERSE>
Jun 25 15:21:04 ev3dev connmand[306]: usb0 {del} route 87.106.208.187 gw 10.42.0.1 scope 0 <UNIVERSE>
Jun 25 15:21:03 ev3dev systemd[1]: Time has been changed
Jun 25 15:21:05 ev3dev ntpd[499]: Listen normally on 2 usb0 10.42.0.3 UDP 123
Jun 25 15:21:05 ev3dev ntpd[499]: peers refreshed
```

Even though it looks like that usb ethernet was activated (usb0 got an IP) - I can't connect. Seem that the system is feezing shortly after.

dlech, 2015-07-03:
> Is the port-1 serial console enabled by default? I am not getting anything to show up there?

Only during early boot. After the kernel modules start loading, the sensor autodetection takes over. The baud rate is 115200. You should always see *something*.

Also, make sure nothing is plugged in during boot. I have seen problems especially when having the USB connected to a host computer during boot. It can cause the EV3 to freeze and it can also cause the host computer to freeze.

dlech, 2015-07-03:
On the initial issue, it sounds like the network connection was interrupted during upgrade. Or maybe the battery died. If you haven’t' written over the SD card yet, you can put it in a computer and restore the files `uImage` and `boot.scr` from the backups in the FAT partition. You will need to run `flash-kernel` manually to get the upgraded kernel installed.

ensonic, 2015-07-03:
After I copy the bak files for `uImage` and `boot.scr` back it boots again and I can log in. Then I run `flash-kernel`:
```
# /usr/sbin/flash-kernel
flash-kernel: installing version 3.16.7-ckt11-5-ev3dev-ev3
Generating kernel u-boot image... done.
Will use /dev/mmcblk0p1 as boot device.
Taking backup of uImage.
Installing new uImage.
find: `/etc/flash-kernel/ubootenv.d': No such file or directory
Generating boot script u-boot image... done.
Taking backup of boot.scr.
Installing new boot.scr.
```
and once it is done reboot. And it gets stuck again :/ Do you have a `/etc/flash-kernel/` directory?

dlech, 2015-07-04:
> Do you have a /etc/flash-kernel/ directory?

Yes. `find: `/etc/flash-kernel/ubootenv.d': No such file or directory` is no problem, it is optional, but we should probably add it (an empty directory) to the flash-kernel package.

Are you trying to upgrade an image from before 1 May 2015? If so, this is not possible. See <http://www.ev3dev.org/news/2015/05/01/Major-Release/>.

ensonic, 2015-07-04:
I am running "ev3-ev3dev-jessie-2015-05-20".

dlech, 2015-07-05:
And you have tried booting with absolutely nothing plugged into the EV3? From you log, I can see that you have Wi-Fi and USB to host computer plugged in. Either one of these could be causing the lock up.

ensonic, 2015-07-06:
Yes, I rebooted without USB plugged. I now also tried without wifi and now it booted. Will try a few more times. Do we have a bug with more details what is causing the lockup?

dlech, 2015-07-06:
So is there a way to make it lock up 100% of the time? If it only locks up when Wi-Fi is plugged in, then I won't be able to reproduce this because I don't have the same Wi-Fi dongle. In a discussion on another issue with @rhempel, I learned that the mainline kernel now has working support for the rtl8188eu driver. Currently we are supplying a separate package for this driver. On the next ev3dev kernel release we will try using the mainline driver, so maybe this will help.

ensonic, 2015-07-06:
Yes, plug the color sensor and it locks-up. You can boot and plug it later and then it locks up. Everytime.

dlech, 2015-07-06:
Well, I'm stumped. Color sensor works for me and no one else has reported this problem.

ensonic, 2015-07-10:
This is working again (kernel update -> 3.35ev3dev5).

closed 2015-07-10 by ensonic