ev3dev/ev3dev #355: NXTCamv4 on ev3dev

hepeti, 2015-07-03:
I have just starting with ev3dev (3.16.7-ckt11-5-ev3dev-ev3) and LEGO, and it seems a amazing and extensive work. As far as I know, the NXTCam device is not supported by ev3dev yet. Would it be hard to use it? I've been reading the "using-i2c-sensors" section, but I didn't find a clear starting point.

Thanks!

dlech, 2015-07-04:
We don't have a ev3dev type driver for the NXTCam yet, but it is still possible to use it using standard Linux I2C libraries. For example @dsharlet has library for stereo vision with two NXTCams [here](https://github.com/dsharlet/ev3cv). If you have any ideas on how we could make it easier to use, be sure to let us know.

hepeti, 2015-07-06:
Thanks for the link! I'll have a look
Regards

burf2000, 2015-10-21:
Has this now been added?  Can EV3Dev talk to PixyCam?

dlech, 2015-10-21:
The "fix committed" label means that the source code had been committed, but it has not been released. I will add another comment and close the issue when it is actually released.

The next kernel release will support automatic detection of the NXTCam, PixyCam with the mindsensors,com PixyAdapter and the PixyCam LEGO edition via the sensor ports. Of course, if you  are clever, you can figure out how to use all of these via the USB port as well or in the case of the PixyCam you can use UART instead of I2C on one of the input ports.

rhempel, 2015-10-21:
I'm getting a PixyCam to help test ...

dlech, 2015-11-09:
Fix released in kernel cycle 8 (`*-8-ev3dev-*`).

closed 2015-11-09 by dlech