ev3dev/ev3dev #356: Allow using speed_sp-like values for regulated mode

WasabiFan, 2015-07-04:
Currently, when one wants to use a motor in regulated mode (to get more accuracy while rotating), they must enable regulation and then specify the `speed_sp` instead of the `duty_cycle_sp`. To a beginning user, this is unintuitive and seems more complex than it needs to be, and requires the programmer to scale the value according to the type of motor being used.

I understand the technical reasons that make this the logical way internally, but could we add a way for the user to not have to worry about this extra complication? I'm imagining another option for the `speed_regulation` called `scaled` that would take a 100 to -100 value and scale it to be in the proper range for the detected motor type. That way, we could have a single property that would work the same way regardless of the regulation mode. I'm not sure which property might take on the other function -- `duty_cycle_sp` seems like the logical choice, but the name "duty cycle" doesn't really apply to the regulated mode like this.

Thoughts?

dlech, 2015-07-05:
> they must enable regulation and then specify the `speed_sp` instead of the `duty_cycle_sp`. To a beginning user, this is unintuitive and seems more complex than it needs to be...

I think that it is equally confusing to use the same attribute for two different types of values depending on the selection of `speed_regulation`. I agree that it is not intuitive, but that is why have documentation. If you want to make this more intuitive, I suggest doing it a the library level. i.e. have two different run functions, `run_unregulated` and `run_regulated` that take duty cycle and speed parameters respectively. 

> and requires the programmer to scale the value according to the type of motor being used.

This is something I think I would like to do something about. On the EV3, 1 tacho motor count is equal to one degree, but on the BrickPi, one tacho motor count is only 1/2 of a degree, so setting the speed setpoint the same on the two different devices will result in a different result.

But I think again, this is going to have to be done at the library level rather than the kernel level primarily because the kernel doesn't do floating point. This way, you don't loose any resolution. On the speed setpoint, having less the 1 degree per second resolution is probably not a big deal. And actually, not that I am thinking about it, having less that one degree resolution on the position and position setpoint is probably not a big deal either, unless someone knows of a motor with slop less than one degree (LEGO motors are about 4 degrees IIRC).

So maybe we do go ahead and do this at the kernel driver level. The `position` and `position_sp` attributes would then have units of degrees instead of tacho motor counts and the `speed` and `speed_sp` attributes would have units of degrees per second instead of tacho motor counts per second. However, the way we have it now, seems more future proof in that it could handle motors that are very different from the ones currently availible, like a motor with a max speed of 1 degree per second.

> I'm imagining another option for the speed_regulation called scaled that would take a 100 to -100 value and scale it to be in the proper range for the detected motor type.

I see what you are getting at here, but I don't think it makes sense to do it at the kernel level. Having two attributes that were dependant on each other doesn't really fit how Linux drivers work. So again, this is something that should be done at the library level. We can expose a `max_speed` attribute so that you can programmatically calculate the speed setpoint as a percentage of the maximum speed.

rhempel, 2015-07-05:
One thing to keep in mind is motors like the Firgelli linear actuators that fake out the driver by looking like regular tacho motors, and then mess up the tacho pulse direction :-)
The `position` and `position_sp` values are in units of pulses, same as the `speed` and `speed_sp`.
Moving to units of degrees puts a kernel level dependency on being able to determine for sure what kind of motor is actually plugged into the port.
My instinct is to move this up to the library level, and leave the kernel with the units of pulses.
Waaay back when, we did have a `duty_cycle_sp` or something like it. The problem was that different motor types have different max speeds. At least with the `speed_sp` you can set the value for the medium and standard EV3 motors and they will both turn at the same speed because they both measure 360 pulses per rev.

dandrus-ev3, 2015-07-25:
FYI, I did some profiling of the EV3 medium and large tacho motors.  It looks like Mindstorms uses a linear equation for power:
for power in [-100, 100]
speed_sp = x * power
medium motor: x = 16.4
large motor : x = 10.3

Note that this maxes out the motors at about 80..
[image]
[image]

dlech, 2015-07-25:
I like it when people do experiments like this. I have some questions though...

* What did you use for a power supply?
* What voltage was the power supply?
* Was there any load on the motor?
* Do the values change when there is a load on the motor?

According to the LEGO firmware (lm21012) source code, the max speed of the large motor is approx. 900 degrees per second and the medium motor is 1200 degrees per second. The conditions here are not specified.

According to <http://philohome.com/motors/motorcomp.htm>, the no load speed at 9V for these motors is 1050 degrees per second and 1560 degrees per second respectively (I converted from RPM). It would be interesting to repeat this experiment with a regulated 9V power supply. I suspect that the curve would not flatten out on the high end.

dandrus-ev3, 2015-07-25:
I wrote a program with the mindstorms widgets that does the following:
for pwr in range (1,10):
  reset motor rotation sensor
  start motor at power 10*pwr
  wait 1 s
  read motor rotation sensor degrees
  wait 1s

I just re-ran it with the power connected and it did affect the top end of both motors but not the rest.  The numbers are with no load on the motor, but I also re-ran it with a light load and it doesn't change.  With a heavier load the numbers drop.

[image]

rhempel, 2016-01-19:
Well, now that I'm working on the tacho motor drivers again, and we have an improved ramping algorithm that handles transitions though 0 properly, it's time to tackle the maximum speed issue again.

The ramping setpoints `ramp_*_sp` describe how long it takes to go from 0 to 100% of the controlling variable. For a DC motor, that means 0 - 100% duty cycle. For tacho motors, that means 0 - 100% of the maximum speed.

Currently, we cap the `speed_sp` value at 2000 to handle the current motors and the possibility of newer high speed motors. But this makes for unpredictable operation when we have different motor types, like the medium and large tachos. They will peak out at different times during a ramp to 2000 pulses per second.

We could have the auto-detect set a `max_speed_sp` (read-only?) that will make it possible to properly scale the ramp times for different motor types.

rhempel, 2016-01-19:
This code in the `legoev3_motor.c` file looks like it will help us to be able to get a maximum speed for a supported motor ID

```
struct lego_device_id legoev3_motor_driver_id_table[NUM_EV3_MOTOR_ID] = {
	LEGO_DEVICE_ID(LEGO_NXT_MOTOR),
	LEGO_DEVICE_ID(LEGO_EV3_LARGE_MOTOR),
	LEGO_DEVICE_ID(LEGO_EV3_MEDIUM_MOTOR),
	LEGO_DEVICE_ID(FIRGELLI_L12_EV3_50),
	LEGO_DEVICE_ID(FIRGELLI_L12_EV3_100),
};
```

dlech, 2016-01-19:
> We could have the auto-detect set a max_speed_sp (read-only?) that will make it possible to properly scale the ramp times for different motor types.

Yes, I think it would be good to expose the max speed from the driver in sysfs. Although leave off the `_sp` suffix since this is not a changeable setpoint.

rhempel, 2016-01-20:
I'll go with the no-load max speed rounded down to the nearest 100 pulses per second - that way we can use the actual limit in the `speed_sp_store` function, instead of the arbitrary 2000

dlech, 2016-04-11:
This fix has been released in the `3.16.7-ckt26-10-ev3dev-ev3` kernel (and other `10-ev3dev` kernels on other platforms). Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

If we did not address all aspects of this issue, please open a new issue.

closed 2016-04-11 by dlech