ev3dev/ev3dev #357: enable cpufreq kernel modules

ensonic, 2015-07-06:
There seems to be a driver for davinci:
http://lxr.free-electrons.com/source/drivers/cpufreq/davinci-cpufreq.c

this could help with PM as discussed in issue #313

dlech, 2015-07-08:
After setting `CONFIG_CPU_FREQ=y`, the kernel fails to compile with:

    /home/david/work/ev3dev2/ev3dev-kernel/arch/arm/mach-davinci/da850.c:1308:2: error: unknown field 'emif_rate' specified in initializer

Apparently, this code has been neglected in the mainline kernel. If anyone wants to look into this some more, I am curious to see what kind of effect it has if we can get it working. I don't have plans to look into it myself though (too many other things to do :wink:).

ensonic, 2015-07-13:
For the cpufreq module, according to
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/arch/arm/mach-davinci/da850.c?id=refs/tags/v4.2-rc2#n1031

you can just remove this line:
https://github.com/ev3dev/ev3-kernel/blob/ev3dev-jessie/arch/arm/mach-davinci/da850.c#L1308

dlech, 2015-07-13:
Like I said, there are lots of other things I would rather be working on, so if anyone want to take some time to try this out and report back the results, we would be most grateful.

ensonic, 2015-07-14:
With removing the line I mention above the cpufreq modules work. Btw. the code comes from the initial kernel source checkin:
commit 0e1e4ddbecbd06cb8da9615e374227ff184e888e
Author: Ralph Hempel <rhempel@hempeldesigngroup.com>
Date:   Tue Oct 1 22:08:53 2013 -0400

    Initial commit of working kernel - does not have all driver support such as Atheros wifi and framebuffer. Realtek wifi works though

I'll post some results shortly.

ensonic, 2015-07-14:
I don't think the cpu-frequency scaling is worth it (on this hardware). Here are the options it enables:
```
# cpufreq-info 
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  driver: davinci
  CPUs which run at the same hardware frequency: 0
  CPUs which need to have their frequency coordinated by software: 0
  maximum transition latency: 2.00 ms.
  hardware limits: 96.0 MHz - 300 MHz
  available frequency steps: 300 MHz, 200 MHz, 96.0 MHz
  available cpufreq governors: userspace, ondemand, conservative, powersave, performance
  current policy: frequency should be within 96.0 MHz and 300 MHz.
                  The governor "ondemand" may decide which speed to use
                  within this range.
  current CPU frequency is 200 MHz (asserted by call to hardware).
  cpufreq stats: 300 MHz:9.01%, 200 MHz:90.99%, 96.0 MHz:0.00%  (1)
```

As you can see, after booting it throttles the clock down to 200MHz. It rarely goes down to 96MHz (in this boot it hasn't clocked down to this level at all).

When looking at the battery-screen I don't see any noticeable difference in the "Power" level. It shows 1.8/1.9W when idle. An external power-meter plugged into the wall-socket measures 2.1W.

If I drop the led-brightness I see a change both on the battery dialog and the external meter
```
for file in /sys/class/leds/ev3-*\:green\:ev3dev/brightness; do echo "16" >$file; done
```

Since any attempt to change the governor or other releated settings locks-up the system, I'd say it is not worth the trouble.

closed 2015-07-14 by ensonic

dlech, 2015-07-14:
Thanks for looking into this. At least we learned something even if the change is not useful to us.

I suppose the `.emif_rate = CONFIG_DA850_FIX_PLL0_SYSCLK3RATE,` line is a carry-over from the LEGO firmware.

ensonic, 2015-07-14:
Jep, I've found that line in a TI patch, but that never made it like that into the kernel. I've been greping my kernel.org checkout and did not find any commit there removing this.